oliexdev/openScale #361: Geonaute Scale 700 (Decathlon scale)

sirio5, 2018-12-08:
[1_btsnoop_hci.log](https://github.com/oliexdev/openScale/files/2659694/1_btsnoop_hci.log)
[2_btsnoop_hci.log](https://github.com/oliexdev/openScale/files/2659695/2_btsnoop_hci.log)
[3_btsnoop_hci.log](https://github.com/oliexdev/openScale/files/2659696/3_btsnoop_hci.log)
[openScale_2018-12-08_13-39.txt](https://github.com/oliexdev/openScale/files/2659697/openScale_2018-12-08_13-39.txt)

[image]

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

oliexdev, 2018-12-09:
please try out this dev version https://github.com/oliexdev/openScale/releases/tag/travis-dev_geonautescale-build

and post the openScale log file (enable it under Settings->About)

sirio5, 2018-12-09:
[openScale_2018-12-09_12-39.txt](https://github.com/oliexdev/openScale/files/2660663/openScale_2018-12-09_12-39.txt)

oliexdev, 2018-12-09:
you are using the wrong dev version see comment above

sirio5, 2018-12-09:
[openScale_2018-12-09_16-49.txt](https://github.com/oliexdev/openScale/files/2660691/openScale_2018-12-09_16-49.txt)

oliexdev, 2018-12-09:
so it is not the same firmware as the Trisa Body Analyze scale. Need to analyse the protocol when I have time.

sirio5, 2018-12-09:
Ok thank you!

brodock, 2023-04-01:
[image]

Got some extra info using BLE Scanner