oliexdev/openScale #362: OpenScale not connecting to Mi Scale v2

ScaleUser, 2018-12-13:
**Describe the bug**
Just purchased the Mi Body Composition Scale Model: XMTZC02HM
Installed OpenScale but it won't connect to the Mi Scale. 

**To Reproduce**
Steps to reproduce the behavior:
1. Install OpenScale
2. Added the bluetooth device from Settings->Bluetooth ("Your Bluetooth Scale")
3. Back to home page
4. Clicked Bluetooth icon to connect to scale. "Connecting to MIBCS" -> No Bluetooth Device Found

Reproduced with latest dev version(https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): 99e3f88

**Expected behavior**
I would expect the App to connect to the Mi Scale and display weight.

**Additional context**
Running a OnePlus 3 

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.
[openScale_2018-12-13_20-25.txt](https://github.com/oliexdev/openScale/files/2675951/openScale_2018-12-13_20-25.txt)

ScaleUser, 2018-12-13:
I powered off my phone. Powered back on again and now it works.

closed 2018-12-13 by ScaleUser

oliexdev, 2018-12-13:
can you test it with this dev version https://github.com/oliexdev/openScale/releases/tag/travis-dev_RxAndroidBle-build

in this version we replaced the whole Bluetooth communication. It should work better but needs more testing.