oliexdev/openScale #368: Lean body mass is not calculated

charleshan, 2018-12-21:
**Describe the bug**
Lean body mass is not calculated.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Overview'.
2. Press Bluetooth button.
2. Measure yourself.
4. See 0.00kg beside Lean body mass.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): Yes.

**Expected behavior**
Shows a number. I'm guessing Weight x Muscle?

Note: I'm using 1byone.

oliexdev, 2018-12-21:
The 1byone doesn't support the calculation of the lean body mass, see the [official homepage](https://www.1byone.com/bluetooth-body-fat-scale-with-ios-and-android-app.html).

Anyway, with openScale you could estimate the lean body mass. Just click in the settings->measurements on the lean body mass and choose one of [these formula](https://github.com/oliexdev/openScale/wiki/Body-metric-estimations).

closed 2018-12-21 by oliexdev

lbdroid, 2021-06-17:
The value it is providing for MUSCLE mass is very close to lean body mass. Just add back the BONE mass and you're there.

In fact, this scale does not actually provide muscle mass, they just incorrectly use that label.