oliexdev/openScale #369: Beta tester needed: Replaced native Android Bluetooth with RxAndroidBle library

oliexdev, 2018-12-21:
I replaced in the [openScale branch RxAndroidBle](https://github.com/oliexdev/openScale/tree/RxAndroidBle) the native Bluetooth with the [RxAndroidBle library](https://github.com/Polidea/RxAndroidBle), see issue #357.

I hope with the new library that some [Android BLE issues](https://github.com/iDevicesInc/SweetBlue/wiki/Android-BLE-Issues) are solved. Two major improvements are:
- the Bluetooth connection should be more stable on all Android smartphones using asynchronous RxJava
- if a Bluetooth connection is lost (whatever it is) a three time retry is performed

Please test the latest [openScale RxAndroid version](https://github.com/oliexdev/openScale/releases/tag/travis-dev_RxAndroidBle-build) with your Bluetooth scale if it works or not (You can also join the [GooglePlay beta program](https://play.google.com/apps/testing/com.health.openscale) for the testing)

Please report here if it works! If you have trouble please attach on this issue the openScale debug log (enable it under settings->about)

Thanks in advance!

lenucksi, 2018-12-23:
I've tested the 1.9-beta (33) from the Google Beta program here. 
OpenScale connects to the scale, however fails to retrieve history data. Upon pressing on the scale the usual "Measured weight" toast appears. 
The retry loop appears to end up in a race condition with an old attempt still running while the new one is initiated eventually leading to complete failure. This also leads to multiple (I assume around 5) toasts overlaying each other leading to illegibility of everything.

I run this on a Galaxy S5 with the latest official LineageOS and a SBF-75 scale.

oliexdev, 2018-12-25:
@lenucksi can your provide the openScale log? Otherwise it is stepping in the dark.

lenucksi, 2018-12-28:
> @lenucksi can your provide the openScale log? Otherwise it is stepping in the dark.

Here's what the "Save diagnostic log" gave me.
[openScale_2018-12-25_14-34.txt](https://github.com/oliexdev/openScale/files/2715391/openScale_2018-12-25_14-34.txt)

oliexdev, 2019-01-10:
@all could you try out again the latest openScale [RxAndroidBle version](https://github.com/oliexdev/openScale/releases/tag/travis-dev_RxAndroidBle-build)

roelf-, 2019-01-11:
Version 1.9-beta (33) almost always failed to connect to my BS444. The new version 1.9-beta (36) works fine again.

oliexdev, 2019-01-11:
@roelf- glad to hear that the latest beta version is working. Thanks for your beta feedback. Do you think it is better to get a message to inform you to step on the scale?

@lenucksi please try out the latest beta version again.

lenucksi, 2019-01-13:
Got the 1.9-beta (36) from Google Play beta program. The scale connects fine and spontaneous measurement toasts are shown too, however no historic data is imported albeit present in the scale.

oliexdev, 2019-01-16:
@lenucksi thanks for the feedback can you try out the latest beta version again I did reimplemented the "pause" function. Could you also add the openScale log file here.

marekz86, 2019-01-19:
I do not connect with the weight Xiaomi Mi Smart Body Scale 2

[openScale_2019-01-19_09-18.txt](https://github.com/oliexdev/openScale/files/2775358/openScale_2019-01-19_09-18.txt)

oliexdev, 2019-01-19:
@marekz86 
your log file logged that you was successful connected to the Mi scale v2 and your data was transfered to openScale?

Anyways the beta has ended. I merged RxAndroidBle into the master tree and release it under version 1.9, so please test it again with the latest release version [openScale 1.9](https://github.com/oliexdev/openScale/releases/tag/v1.9)

closed 2019-01-19 by oliexdev