oliexdev/openScale #372: not receiving data from Yunmai mini

SarahDal, 2019-01-01:
My Samsung Galaxy S9 pairs with my Yunmai and I am able to add the device in OpenScale, however beyond that I can't get any data from the scale. I feel like I've missed a step! the scale speaks to the Yunmai app no problem.

I have reinstalled openscale, rebooted phone, uninstalled the Yunmai app.

erijo, 2019-01-01:
Have you clicked the Bluetooth icon in the app's main view? That's needed to make the app connect to the scale and fetch measurements.

SarahDal, 2019-01-01:
Thank you for your quick reply, and happy new year! There doesn't appear to be a bluetooth icon in the main view.

[image]

erijo, 2019-01-01:
For some unknown reason the button is sometimes black. Try clicking just left of the "+" button.

SarahDal, 2019-01-01:
yep that's solved it - thank you!

closed 2019-01-01 by SarahDal