oliexdev/openScale #373: Scale support for QN-Scale

vanadario, 2019-01-06:
Hi,

I was happy to see that I can connect my QN-Scale from [Yolanda](http://www.yolanda.hk/#) to openScale since version 1.8.4:
[image]

Unfortunately only the weight is shown.
Additionally the weight seems to be read several times while the measuring is still running and that results in several entries for one measurement procedure in the table in openScale.

I would like to mention at the beginning that the vendor offers to open the API:
[http://www.yolanda.hk/en/customization.html](http://www.yolanda.hk/en/customization.html)
Perhaps it is the easiest way to ask for help there?

I followed step 1 and 2 [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale) anyway but wasn't very successful with step 3.
I was able to identify the weight in all logs, I will mention the package no. with every measurement. The weight is in byte 15 and 16, multiplied with 100 in big endian.
But the weight seems to be transferred repeatedly in almost identical packages during the measurement, probably explaining that there are several entries in the table of openScale.
But I was not able to identify data like height or sex in the packages transferred to the scale, so the BMI and fat seems to be calculated in the vendor's app. But I was also not able to identify something looking like the impedance in the packages transferred from the scale. But it must be sent after these several identcal packages with the weight, because openScale already shows these entries and only after this the scale seems to measure the impedance.

Data from step 1:
1. measurement:
[btsnoop_hci_1.log](https://github.com/oliexdev/openScale/files/2730755/btsnoop_hci_1.log)
[image]
Package No. 49

2. measurement:
[btsnoop_hci_2.log](https://github.com/oliexdev/openScale/files/2730756/btsnoop_hci_2.log)
[image]
Package No. 49

3. measurement:
[btsnoop_hci_3.log](https://github.com/oliexdev/openScale/files/2730757/btsnoop_hci_3.log)
[image]
Package No. 43

4. measurement (different person):
[btsnoop_hci_4.log](https://github.com/oliexdev/openScale/files/2730758/btsnoop_hci_4.log)
[image]
Package No. 39

Data from step 2:
1. measurement:
[openScale_2018-12-22_22-24.txt](https://github.com/oliexdev/openScale/files/2730762/openScale_2018-12-22_22-24.txt)

2. measurement:
[openScale_2018-12-22_22-58.txt](https://github.com/oliexdev/openScale/files/2730763/openScale_2018-12-22_22-58.txt)

Here are pictures of the scale, eventually for completing your wiki when it should have been possible to implement the scale:
front:
[image]
back:
[image]

Thank you in advance for trying to support this scale!

oliexdev, 2019-01-07:
thanks for your log files, if we have more we can test better. Unfortunately, we have to reverse engineer the `libyolanda_calc.so` library. Please read issue #291 for further information.

dockerss, 2019-01-08:
See - https://github.com/oliexdev/openScale/issues/291#issuecomment-452367604

vanadario, 2019-01-09:
Thanks for your replies.
I will test the QNScaleDemo-1.0.apk in the next days.

oliexdev, 2019-01-11:
@vanadario could you provide for measurement 1 and 2 which user you had entered? (sex (male or female), age and body height in cm)

vanadario, 2019-01-12:
@oliexdev I entered a testuser named "Horst", male, day of birth 02.05.1981 (tt.mm.yyyy, so currently 37 years old), height 178 cm.

I just also tried to install the QNScaleDemo-1.0.apk but it could not be installed without a clear error message. I'm afraid the phone I#m using for these tests is no longer supported, (HTC One mini2 with Android 4.4.2).
My current phone is owned by my company. When I activate sideloading of apps there the KNOX-workspace would be deleted imediately, so I can't try it with this one...

oliexdev, 2019-01-16:
@vanadario did you use the same user information for both measurements?

vanadario, 2019-01-16:
@oliexdev Yes, for measurements 1, 2, 3 I used the same.
For measurement 4 I used another one, female, 176 cm, 33 years old.

vanadario, 2019-01-24:
@oliexdev Since the update to version 1.9 (37) it works with my QN-Scale, values are no longer transferred repeatedly and also body fat etc. are displayed.
Thank you very much!
The values for body fat etc. differ from the values in the vendor's app, but I assume that's just due to the estimation formula the app uses? For me the value itself is not the relevant but the trend.

[image]
[image]

dockerss, 2019-01-30:
@vanadario what is the vendor app? Can you share a link?
I was getting a body fat reading of +-2% when compared with fitindex app.
You can look at the logs for
 resistance1: 
 resistance2: 
and share the values

vanadario, 2019-02-01:
> 
> 
> @vanadario what is the vendor app? Can you share a link? Is it this one http://www.yolanda.hk/mobile/apps/oversea_app_down

Your link doesn't work now. I use this one:
https://play.google.com/store/apps/details?id=com.yolanda.foreign&hl=en_US

> I was getting a body fat reading of +-2% when compared with fitindex app. https://play.google.com/store/apps/details?id=com.qingniu.fitindex
> You can look at the logs for
> resistance1:
> resistance2:
> and share the values

I compared Fitindex, Yolanda and openScale. Values in Fitindex and Yolanda are almost the same, it even seems to be the same app with a different theme. Screenshots below.

Can you give me a hint where I can find the logs? I even can't find a data folder for Yolanda or Fitindex in Android/data.

Fitindex:
[image]

Yolanda:
[image]

openScale:
[image]

srwareham, 2019-03-08:
Can this issue be closed? 

If so, I believe the wiki can be updated with a new scale "QN-Scale" with the image above and details identical to the FITINDEX ES-26M one in #434

oliexdev, 2019-03-08:
@srwareham well I don't think so. Currently, the QN-Scale uses the Trisa Body Analyse library but somebody has to be reverse engineered the original `libyolanda_calc.so` library to get the correct values of the body measurements.

Aypac, 2020-02-15:
It appears to me like this exact scale is sold under many vendor names. I have a version from "RENPHO" (same model number though).
 <details>
  <summary>Show/Hide pictures</summary>
  [image]
[image]
[image]
</details> 

As a result in the [propriety app](https://play.google.com/store/apps/details?id=com.qingniu.renpho&hl=en) I get

[image]
while I get
[image]
from this app.

 <details>
  <summary>Show/Hide raw data</summary>
"biceps","bone","caliper1","caliper2","caliper3","calories","chest","comment","dateTime","fat","hip","lbm","muscle","neck","thigh","visceralFat","waist","water","weight"
0.0,4.786194,0.0,0.0,0.0,0.0,0.0,"","2020-02-15 17:06",15.9795,0.0,0.0,45.815155,0.0,0.0,0.0,0.0,61.873444,70.4
</details>

I'll try and post some btsnoop data later, if that would help.

mzzntn, 2020-10-29:
I also have a scale that appears on Bluetooth devices as QN-S500. The scale is sold by Kamtron, in my particular case.

If you need any data or information to be able to support it, I'm here to help :)

fooxax, 2021-02-07:
Hey all, 
Just to add another scale sold as "Hutbit" and identified as QN-Scale. The official app is called feelfit but as mentioned earlier seems to be similar to other apps which are linked to QN-Scales.

[image]
[image]

The most interesting things to know,  besides of what already works,  are body fat and muscle mass.  Will see if I can find a matching formula.

gammerboy, 2021-09-28:
It seems the same scale is sold as Actofit smartscale lite in India. Which showed up as QN-scale under bluetooth but didn't connect in their own app. Not sure if they are using a different firmware in their device and I got one which did not got pushed out with the manufacturer firmware.

gammerboy, 2021-09-28:
It looks exactly the as the photo posted by the op. Openscale crashes when trying to connect to it. I'm not sure if it has some other firmware running on it tho. 

`Build version: 2.3.2-dev_5d697fe9_2020-11-05 
Build date: 1981-01-01 01:01:02 
Current date: 2021-09-28 19:00:38 
Device: OnePlus ONEPLUS A3003 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at java.util.Objects.requireNonNull(Objects.java:245)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1192)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:200)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:188)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:460)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:462)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:462)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:173)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:329)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:381)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:223)
	at android.app.ActivityThread.main(ActivityThread.java:7664)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
`