oliexdev/openScale #376: App crashes on Chart tab

Disasm, 2019-01-12:
**Describe the bug**
App crashes on Chart tab.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Settings->Chart.
2. Enable Regression weight line, set polynomial degree to 5 (I don't remember the exact value).
3. Wait for the beginning of 2019.
4. Add measurement.
5. Go to Chart tab.
6. Observe crash.
7. Restart application.
8. Observe crash.

Reproduced with 1.8.4 (build from F-Droid)

**Debug log**
```
Build version: 1.8.4 
Build date: 1979-11-30 03:00:00 
Current date: 2019-01-12 14:59:06 
Device: Xiaomi Redmi 4X 
 
Stack trace:  
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.health.openscale/com.health.openscale.gui.MainActivity}: java.lang.ArrayIndexOutOfBoundsException: length=5; index=5
 at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2778)
 at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2856)
 at android.app.ActivityThread.-wrap11(Unknown Source:0)
 at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1589)
 at android.os.Handler.dispatchMessage(Handler.java:106)
 at android.os.Looper.loop(Looper.java:164)
 at android.app.ActivityThread.main(ActivityThread.java:6494)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:440)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
Caused by: java.lang.ArrayIndexOutOfBoundsException: length=5; index=5
 at com.health.openscale.core.utils.PolynomialFitter.gj_swap(PolynomialFitter.java:186)
 at com.health.openscale.core.utils.PolynomialFitter.gj_echelonize(PolynomialFitter.java:156)
 at com.health.openscale.core.utils.PolynomialFitter.getBestFit(PolynomialFitter.java:118)
 at com.health.openscale.gui.fragments.GraphFragment.generateLineData(GraphFragment.java:447)
 at com.health.openscale.gui.fragments.GraphFragment.access$1100(GraphFragment.java:79)
 at com.health.openscale.gui.fragments.GraphFragment$chartTopValueTouchListener.onValueSelected(GraphFragment.java:588)
 at lecho.lib.hellocharts.view.ColumnChartView.callTouchListener(ColumnChartView.java:73)
 at lecho.lib.hellocharts.view.AbstractChartView.selectValue(AbstractChartView.java:449)
 at com.health.openscale.gui.fragments.GraphFragment.generateColumnData(GraphFragment.java:528)
 at com.health.openscale.gui.fragments.GraphFragment.generateGraphs(GraphFragment.java:563)
 at com.health.openscale.gui.fragments.GraphFragment.updateOnView(GraphFragment.java:250)
 at com.health.openscale.core.OpenScale.registerFragment(OpenScale.java:599)
 at com.health.openscale.gui.fragments.GraphFragment.onCreateView(GraphFragment.java:228)
 at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2439)
 at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManager.java:1460)
 at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1784)
 at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManager.java:1852)
 at androidx.fragment.app.FragmentManagerImpl.dispatchStateChange(FragmentManager.java:3269)
 at androidx.fragment.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManager.java:3229)
 at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:201)
 at androidx.fragment.app.FragmentActivity.onStart(FragmentActivity.java:620)
 at androidx.appcompat.app.AppCompatActivity.onStart(AppCompatActivity.java:178)
 at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1334)
 at android.app.Activity.performStart(Activity.java:7029)
 at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2741)
 ... 9 more
```

oliexdev, 2019-01-14:
Thanks for reporting. Unfortunately, I can't reproduce this error. Can you test maybe [this](https://github.com/oliexdev/openScale/releases/tag/travis-dev_MPAndroidChart-build) version? 
We replaced in this version the Chart library. Maybe it is already fixed in this branch?

closed 2019-01-28 by oliexdev

Disasm, 2019-02-14:
Recovered from a backup and still can't reproduce this. Sorry.