oliexdev/openScale #377: [FR] Mood tracker

OffifialMITX, 2019-01-13:
Would it be possible to include a mood tracker that can be used mixed or separately from the app's current main functionality? I am looking for a F(L)OSS app with that functionality, but couldn't find anything yet, and from what OpenScale does, it appears that this wouldn't require any big changes in the code.

I understand though, if this is not what you want to achieve and you wouldn't want to do that.

erijo, 2019-01-13:
You could use emojis in the comment field for this.

OffifialMITX, 2019-01-13:
Of course, but I'm looking for something that can show me statistics, and I'd like it if it w

oliexdev, 2019-01-14:
It is not the goal of openScale to track the mood of the user.
You could learn to program your own Android app and create an open-source Mood tracker app. But I see you did also ask other projects to implement this feature. Maybe they are willing to do it.

closed 2019-01-14 by oliexdev