oliexdev/openScale #378: Honor CH100 scale support

anongeek, 2019-01-13:
Hello,
I'm looking for support for Honor CH100. I do some capture
Hope you could help for integrating this scale into the application.
I've attached the openScale logs and also HCI snoop.

Thanks in advance !

Best regards,
[image]

[openScale_2019-01-13_23-00.txt](https://github.com/oliexdev/openScale/files/2753597/openScale_2019-01-13_23-00.txt)

[hci_snoop20190113225006.cfa.txt](https://github.com/oliexdev/openScale/files/2753598/hci_snoop20190113225006.cfa.txt)

oliexdev, 2019-01-14:
Thanks we need all true data of your measurement regarding to the hci_snoop log for e.g. fat, water, muscle mass, sex, body height, activity level and so on.

anongeek, 2019-01-14:
Hello, 

here's the data requested for the current measure :
sex : male
weight : 66,3kg
fat percentage : 18,8%
muscle mass : 51,3kg
I stay available if you need more informations.

best regards,

jpintor14, 2019-01-17:
Hello, I'm working in this integration too.

According a Huawei Support ticket, the Body Fat Scale uses the "Weight Standard Profile". The specifications doc [here]( https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=293525).

I see that if the scale support multiple users, to read the "Weight" or "Body Composition Measurement", you have to write the "User Control Point" first. The document says (setcion 4.6.3.2.2):

> To request the consent of a Weight Scale user in order to access their UDS Characteristics, the Collector shall use the Consent Op Code followed by an array of 3 UINT8 Parameter Values that represents the User Index (1 octet) followed by the Consent Code (2 octets) defined by the user as defined

But, How I know what is this "Consent Operator Code".

Best regards,
Javier Pintor

oliexdev, 2019-01-17:
@anongeek your entered body height is missing. Could you provide this piece of information.

@jpintor14 is your implementation open-source? You could maybe look into the hci_snoop log to find the "consent operator code", see [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale).

jpintor14, 2019-01-18:
Thanks for the help! This is the [log](https://github.com/oliexdev/openScale/files/2772939/btsnoop_huawei.log) of the official app.

I can't see in it the write operation with the standart command(0x2A9F). 
There are two "Sent Write Request" with the characteristic 0xFAA2.
I can't decode it yet, any idea?

This is a piece of the code of my project (Xamarin Android App in C#):

   **- DeviceControlActivity:**

public void LoadPesoHuaweiCharacteristic(BluetoothGattService gattService)
{

    if (UUID_USER_DATA_SERVICE.ToString() == gattService.Uuid.ToString())
    {
        BluetoothGattCharacteristic userControlPoint = null;
        BluetoothGattCharacteristic userIndex = null;

        IList<BluetoothGattCharacteristic> gattCharacteristics = gattService.Characteristics;
        foreach (BluetoothGattCharacteristic gattCharacteristic in gattCharacteristics)
        {
            if (UUID_USER_INDEX.Equals(gattCharacteristic.Uuid))
            {
                userIndex = gattCharacteristic;
            }
            if (UUID_USER_CONTROL_POINT.Equals(gattCharacteristic.Uuid))
            {
                userControlPoint = gattCharacteristic;
            }
        }

        if (userIndex!=null && userControlPoint != null)
        {
            Thread.Sleep(300);
            mBluetoothLeService.SetCharacteristicNotification(userIndex, true);
            Thread.Sleep(300);
            mBluetoothLeService.SetCharacteristicNotification(userControlPoint, true);
            Thread.Sleep(300);
            mBluetoothLeService.ReadCharacteristic(userIndex);
            Thread.Sleep(300);
            mBluetoothLeService.ReadCharacteristic(userControlPoint);
            Thread.Sleep(300);

            mBluetoothLeService.mandado = true;
            mBluetoothLeService.setDeviceControlActivity(this);

            Intent gattServiceIntent = new Intent(this, typeof(BluetoothLeService));
            BindService(gattServiceIntent, mServiceManager, Bind.AutoCreate);
        }

    }

    if (UUID_WEIGHT_SCALE_SERVICE.ToString() == gattService.Uuid.ToString())
    {
        BluetoothGattCharacteristic weightMeasurement = null;

        IList<BluetoothGattCharacteristic> gattCharacteristics = gattService.Characteristics;
        foreach (BluetoothGattCharacteristic gattCharacteristic in gattCharacteristics)
        {
            if (UUID_WEIGHT_MEASUREMET_HUAWEI.Equals(gattCharacteristic.Uuid))
            {
                weightMeasurement = gattCharacteristic;
            }
        }

        if (weightMeasurement != null && weightFeature != null)
        {
            mNotifyCharacteristic = weightMeasurement;
            Thread.Sleep(300);
            mBluetoothLeService.SetCharacteristicNotification(weightMeasurement, true);
[... 145 lines cut ...]
        mBluetoothGatt.WriteDescriptor(descriptor);
    }
    #endregion
}

public void BroadcastUpdate(String action, BluetoothGattCharacteristic characteristic)
{
    Intent intent = new Intent(action);

    byte[] data = characteristic.GetValue();
    Log.Info("SendBroadcast", characteristic.Uuid.ToString() );
    for (int x = 0; x < data.Length; x++)
    {
        Log.Info(characteristic.Uuid.ToString(), x + ": " + data[x]);
    }
    

    SendBroadcast(intent);
        
}

 The UpdateBroadcast() method is not fired with the rightcharacteristics

Best regards,
Javier Pintor

oliexdev, 2019-01-18:
what is happened if your send the same 0xFAA2 command to the scale?

raven-ie, 2019-01-22:
I just got Huawei Body Fat Scale AH100 and wanted to open ticket along with the logs - is Honor CH100 just a rebrand of the same scale? If so I could maybe help with some logs?

jpintor14, 2019-01-22:
@raven-ie yes, is the same model.
I have the Huawei AH100, but later de Bluettoth device is called CH100 too.

@oliexdev I am still testing writing to the 0xFAA1, but the FxFAA2 is not changed.
I recevive the FxFAA2 values when the Scale connect to the App, not when the measurement is completed

jpintor14, 2019-01-23:
There are a point that the app receive the same structure of the characteristic faa2.
I have repeated some times the mesurement and this is the results, but ? can't decode yet.
The dynamic part is in parentheses:

bc 11 0e bf (c6 c3 7e) d7 d6 eb d5 (fe 21 c2 11 8a 18) 76 0b => 84.7 kg
bc 11 0e bf (dd c3 78) d7 d6 eb d5 (fe 23 cf 19 8a 56) 76 0b => 85.2 kg
bc 11 0e bf (dd c3 78) d7 d6 eb d5 (fe 23 f9 37 8a 52) 76 0b => 85.2 kg
bc 11 0e bf (da c3 67) d7 d6 eb d5 (fe 24 d4 2b 8a 51) 76 0b => 85.1 kg
bc 11 0e bf (da c3 67) d7 d6 eb d5 (fe 24 d6 11 8a 50) 76 0b => 85.1 kg
bc 11 0e bf (dc c3 78) d7 d6 eb d5 (fe 24 c0 2f 8a 5f) 76 0b => 85.3 kg
bc 11 0e bf (dd c3 67) d7 d6 eb d5 (fe 24 c4 37 8a 5e) 76 0b => 85.2 kg
bc 11 0e bf (dc c3 67) d7 d6 eb d5 (fe 24 f3 09 8a 5d) 76 0b => 85.3 kg
bc 11 0e bf (da c3 66) d7 d6 eb d5 (fe 24 fe 16 8a 5c) 76 0b => 85.1 kg
bc 11 0e bf (da c3 64) d7 d6 eb d5 (fe 25 cf 38 8a a0) 76 0b => 85.1 kg

The second variable part could be a datetime, but the first one, it doesn't have much logic:

da c3 =? 85.1 kg
dd c3 =? 85.2 kg
dc c3 =? 85.3 kg

These values of 0xfaa2 are received when the app writes 'bd 01 20 21'

david-gam, 2019-04-18:
Hola, estoy intentando leer las lecturas de peso de la bascula Huawei ch100 y ch18 y estoy teniendo los mismos problemas, alguien puede ayudarme?

jpintor14, 2019-04-19:
I could research and the info is encrypted, and the Huawei support answer me that the product not allow third party integrations

quixony, 2019-06-01:
Is any progress been made within this issue? I could provide additional decoding data if that helps.

jahav, 2020-06-03:
Hi,

I have taken several samples from my scale (Huawei AH100, but reports as ChipseaT_82:65:40 (CH100S), same as other ogs from the thread) and analyzed the data to determine an algorithm to get a weight from the raw value of the bluetooth log. 

Unfortunately it gives correct values only for my scale (on my sample of 60 values). When I tried to use same algorithm for other logs posted in this thread, it doesn't give expected values.

Basically I have to find a received value with characteristic 0xfaa2 starting with **bc110e** (it seems to always be followed by another that starts with **bc118e**, not sure what is it for yet). The fifth and sixth byte are a raw value for weight in little endian.

Example:
* raw value from log: bc 11 0e 30 **6e 59** 8d e8 b3 b5 10 44 6e 58 66 00 7e 42 ce
* the app shows 83.0 kg.

You can find it easily in the bluetooth dump, there is always a second received value that looks similar, it has similar prefix.

Ok, we have a raw value, it's in little endian: 596e. 
Convert it to binary: `0101 1001 0110 1110`
Split it to following parts, dump rest
* bits0-4: b01110    (14)
* bits5-6: b11   (3)
* bits7-8: b10   (2)
* bit9: b0   (0)

Now we transform pieces of the raw value into kg value

```csharp

decimal TransformRawToKg(int raw)
{
	int bits0_4 = raw & 0b00_0001_1111;
	int bits5_6 = (raw & 0b00_0110_0000) >> 5;
	int bits7_8 = (raw & 0b01_1000_0000) >> 7;
	int bit9 = (raw & 0b10_0000_0000) >> 9;

	var kg_value = 
	      (((bits0_4 + 16m) / 10m) % 3.2m)    // 1
	    + 3.2m * (                            // 2
		           ((bits5_6 + 2) % 4)    // 3
	             + (bits7_8 << 2)             // 4
	             + ((~bit9 & 1) << 4)         // 5
	             );
	 return kg_value;
}

```

Line explanation
1. First 5 bits are a part of weight that don't really need much adjustment. When bit0_4 go up by 1, weight goes up by 0.1 kg. The modulo and scale are for alignment.
2. Since bit0_4 specify values less than 3.2, the upper bits represent a band 3.2 kg wide, so normalize the transformed value.
3. bits5_6 are kind of weird. The bit values cycle correctly 10 -> 11 -> 00 -> 01, but the carry that should happen when 11 increases to 00 + carry doesn't happen. Instead, the carry happens when 01 goes to 10. The modulo basically shifts the cycle so carry is in correct place (i.e. b10 changes to b00, b11 changes to b01, b00 to b10 and b01 to b11).
4. bits7_8 are fine as is, no need to change them
5. bit9 is 1 until roughly 55 kg and then it is 0. So just invert it, so it is 0 for values under roughly 55 kg and 1 for higher values.

The raw values from my logs (along with code for checking the algorithm) can be found at [pastebin](https://pastebin.com/g3Qsn3F6).

I have attached a some of the logs I used for the decoding:
[openScale-AH100-logs.zip](https://github.com/oliexdev/openScale/files/4726426/openScale-AH100-logs.zip)

jahav, 2020-06-06:
These weird shifts are xor, so no need for the transformation log from my previous post, you just need a correct xor value (not sure how to get it) and xor it. For me, it's 0x5a50. For @jpintor14 , its' 0xC089.

Take the raw value, xor it and you have the result in decagrams.

Example:

> da c3 =? 85.1 kg
> dd c3 =? 85.2 kg
> dc c3 =? 85.3 kg

xor value is = 0xc3da ^ 851 = 0xC089;
0xc3da (raw value) ^ 0xC089 = 851;
0xc3dd ^ 0xC089 = 852;
0xc3dc ^ 0xC089 = 853;

diegoalbertopp, 2020-07-06:
Hello, was it possible to integrate this scale or is someone trying? Sorry for my English.

jahav, 2020-07-06:
@diegoalbertopp I gave up.

diegoalbertopp, 2020-07-09:
Can I help with something? I have this weight scale and am interested in this, but I don't even know how to start :(

quixony, 2020-08-13:
Might that XOR value be related to SerialNumber (back of scale), or be calculated while the pairing process? Maybe from firmware and SN or something like that? I would like to start a colletion of XOR-values with transfered data while paring to maybe get the calculation for that XOR value. 
@jahav do you think there is a chance to crack that algorithm?

call-me-matt, 2020-11-19:
Hi there. I am also very interested in this - especially since the Huawei app for the AH100 cannot export its values (so everything is lost when you get a new phone).
@diegoalbertopp did you give up only because of the differing constant, or where there other issues coming up? If it is only the constant that is different for each user, maybe the displayed weight can be entered on first coupling and by this be determined?

jahav, 2021-05-08:
I have made effort to work on a decoder it and later abandoned and it would be shame not to share it (Directive 2009/24/EC, Article 6). Most of the stuff is not verified, there might be errors.

# OpenScale - Communication 

Receiver receives BlueTooth packets that are processed. There are three kinds of packets:
* Notification - scale notifies device that something has happend. There is a ~20 types of notifications (e.g. scale woke up, fat measured, history uploaded, reminder set, units changed, ). Notification packets start with a value `0xBD`.
* Chunks - a single notification is split into two packets. Chunks has to be combined and decrypted to get notification packet. Chunk packets start with a value `0xBC`.
* SDK auth result packet - not sure what is it for, if you receive [0x05,0xFA, ...], it means SDK authorized packet. Mostly ignored in further text. The packets has a specific pattern.

Each packet consists from a header 3 bytes long and payload. When the payload is referenced in the document, payload[0] is the first byte of payload, not first payload of packet (`payload[0] = packet[3]`).

Header is used to detect what kind of packet it is and how to process it before passing the packet data into notification parser.

# Cipher and keys
In some cases, the payload of a command and notification packets is encrypted and it must be decrypted before sending or decrypted before processing. Both encryption and decryption use same key and iv and other parameters.

The protocol is using AES cipher with CTR mode. CTR mode effectively means that AES creates a stream of bytes from key+iv and then XORs them with plaintext payload bytes.

The cipher is initialized with same key+iv for each packet, so effectively each packet is XORed with same magic byte array.

Key for AES is created from scaleKey (a key created when pairing the scale to the device, various length) and preset key (const for all occurances). Key is created by taking initialKey and replacing its first bytes with scaleKey obfuscaled by scale MAC address. IV is also preset (const for all occurances) and is not modified.

```
Preset:
initialKey: 3D A2 78 4A FB 87 B1 2A 98 0F DE 34 56 73 21 56
iv: 4E F7 64 32 2F DA 76 32 12 3D EB 87 90 FE A2 19

Example: 
scaleMac:         12 34 56 78 9A BC
scaleKey:         DE F0 12 34 67 89 AB
obfScaleKey(XOR): CC C4 44 4C FD 35 B9  (last byte is XORed with 12, because scaleMac is repeated)

key: CC C4 44 4C FD 35 B9 2A 98 0F DE 34 56 73 21 56
```
The obfuscated scale key is also sent as the first command packet type `0x24` to the scale during authentication.

# Receiving data from the scale
Received packet can be either a notification packet (kind `0xBD`) or can be a chunk of encrypted packet (kind `0xBC`).

Notification packet should have its payload deobfuscated and processed.

Encrypted packet is wrapped in two chunk packets and should be assembled, decrypted, deobfuscated and processed. There are two encrypted packets: type `0x0E` (measurement result) and and `0x10` (history).

## Structure
Notification packet contains information from the scale about a result of some action. Each packet consists from a three byte header:
* kind of packet - is it a notification packet (`0xBD`) or a chunk (`0xBC`).
* payload length - length of (decrypted) payload.
* type - determines what kind of notification is it, what kind of payload to expect. Second chunk has a type of first chunk + 0x80.
and optional payload.

Packets that don't fit the expected structure should be discarded.

## Assembly
Encrypted packet consists from two chunks and can only be encrypted once both are received. Each chunk is either
* First chunk - its type (header[2]) is a notification type of the packet that was encrypted.
* Second chunk - its type (header[2]) is  notification type of the packet that was encrypted shifted by `0x80`.

Example: encrypted measurement packet (type of measurement packet is `0x0E`) has following header
```
First chunk> BC:11:0E:encryptedPayload1
Second chunk> BC:11:8E:encryptedPayload2
```

Length specified in the chunks is not the length of the encrypted payload, it should be same as the length of decrypted and assembled payload.

Since encrypted packet is wrapped in two chunk packets, there is a chance of missing one chunk packet or other.

## Decryption
Once both chunks are received, the payload of both chunks should be individullay decrypted using *AES* cipher with *CTR* mode and *PKCS7* padding.

[... 312 lines cut ...]
```
**TODO**: Impedance is 200-1500 range and if outside, both impedance bytes at 0. What is sex 0 or 1? Decrypt.
Android_id https://developer.android.com/reference/android/provider/Settings.Secure#ANDROID_ID (basically an identifier for the lifetime of device... unless you factory reset).

## Get history command 0x0B
Send me a notification 0x10 with records about past measurements stored in teh scale (even ones taken offline). Once all records are sent, finish with notification 0x19.

# Decoding example

You have a wireshark dump and want to see measurements:
* Open wiresharp BT dump and display only relevant packets:
```
btatt.handle == 0x36 or btatt.handle == 0x38 
```
* Add a column for value in Wireshark (Title:Value, Type:Custom, Fields: `btatt.value`).
* Find authorization in the dump: It is a packet where value starts with DB0824[cipherKey].
* Create a *key* by combining cipherKey with preset initialKey (see *Cipher and keys*)
[... cut ...]
* Decrypt `{payload1}` and `{payload2}` using the cipher with the *key* and *initialIV*.
* Deobfuscate the `{payload1}{payload2}` using the scale MAC
* Decode the payload according to structure from notification type `0x0E`

I created a gist of my test program I used during research (horrible, but should provide tips about how to decrypt a packet, it explanation wasn't clear) with values from the first post by @anongeek: https://gist.github.com/jahav/da3e46194f0bff6d5b96aa9f12e616c8

@diegoalbertopp FYI

marcovrv, 2021-11-16:
Hi, it would be great if this scales are implemented, given that Huawei is no longer updating it's apps on Google Play. Also because even Huawei Health app can only export data in picture format, not text that may be pasted.

Iliaroz, 2022-07-31:
Hello!

I have wrote some code, basing on @jahav  description. 
Could someone who has the scale test my code?
it placed on my github (branch `AH100`)
https://github.com/Iliaroz/openScale.git

It should read online measurements and probably offline.

RazvanSt, 2022-08-01:
Thank you for the patch.. I've tested it - screenshot attached.

It seems that are no readings for Total Body Water, Muscle and the date is wonky.

[image]

quixony, 2022-08-01:
I'd really like to test the version as well, but I am not familiar with building the apk. Could someone provide it to me? Especially I have a CH100 skale (so not the same as RazvanSt: AH100).

Iliaroz, 2022-08-01:
Thank you for testing..

@RazvanSt 
The scale does not send muscle, water and bone info. So, it should be calculated somehow... in my todo list.
wonky Date is corrected, thank you.
Please, could you also check, does it read offline measurements? ( I mean, you weigh yourself without OpenScale app and after few minutes again with app)

@quixony 
Please, find apk file on my github, branch `AH100-TEST` in folder `test_apk`
Please, could you also check, does it read offline measurements? 
( I mean, you weigh yourself without OpenScale app and after few minutes again with app, in measure table should be both records)

RazvanSt, 2022-08-02:
Unfortunately there is no offline measurements reading.  I only see the one that is taken while the application is connected to the scale.

quixony, 2022-08-02:
I can confirm that. No offline measurements are shown.
 
Online measurements work great (besides the wrong date, and the app crashes if gps is disabled on adding Bluetooth devices).

I'd really appreciate if these scales get supported by OpenScale (main).

Iliaroz, 2022-08-07:
Please, check last version. Does it read offline measurements? (Weight yourself online first)

ps. Water, muscles, bones are still in progress.

quixony, 2022-08-07:
Did you check the apk? To my system it seems broken and cannot be installed :-/

Additionally: could you explain in detail why the measurement should be performed online, when offline data transfer should be checked?

Iliaroz, 2022-08-09:
@quixony 
Please, try again.

When you connect app to scale, your user data (age, height, weight) and user-key send in encrypted form to scale. when you make offline measurements, scale checks which user this measurement apply to (basing on weight). Next time app connected to scale, offline measurements bound to provided by app key read from scale. 
In this case due to error in program, two different keys was used (so, no offline measurements)

RazvanSt, 2022-08-10:
I've tested the new version - no joy. No offline data.

Iliaroz, 2022-08-10:
@RazvanSt 
Ha-ha. Nice ;(
Could you send me bluetooth packet log? (the same as for new scale request) ???

RazvanSt, 2022-08-10:
Here are the logs:
[openScale_2022-08-11_01-14.txt](https://github.com/oliexdev/openScale/files/9304422/openScale_2022-08-11_01-14.txt)

There was one offline reading on the device, but it does not appear in app.

Iliaroz, 2022-08-13:
@RazvanSt 
Thank you for your trying and for your debug info.
I have a few questions to you:
1. do you have `ch100` or `huawei health` app in use?
2. if yes, did your offline measure appear in this official app?
3. when you made your offline measurement, was on scale shown fat percentage or "--" sign?

and... can you try again?

RazvanSt, 2022-08-13:
1. I have Huawei Body Fat Scale app installed
2. No, I have no offline measurements also in that app
3. Measurements showed fat percentage also.

I am not at home for the next two weeks, so no way to measure again.

rufik, 2022-08-26:
My weight is floating no more than +/- 1.0kg from initial one. So close enough, right? :)
I'm gonna try to make a few (2-3) offline measurements and then make online one. Should I save debug log and publish here? Would it be helpful?

Iliaroz, 2022-08-26:
Scale documenatation says, that weigth could be +/- 2 kg.. 
Yes, it would be nice to have log, thank you.

rufik, 2022-08-26:
I did 2 offline measurements and then third one (at 11:33) as online. As expected, only the online one has been fetched successfully.
See the log attached: [openScale_2022-08-26_11-33.txt](https://github.com/oliexdev/openScale/files/9432134/openScale_2022-08-26_11-33.txt)

Iliaroz, 2022-08-26:
Could you repeat? again make a few offline  measurements and online one

rufik, 2022-08-26:
Well, pretty the same story - only online measurement.
[openScale_2022-08-26_17-25.txt](https://github.com/oliexdev/openScale/files/9434950/openScale_2022-08-26_17-25.txt)

Iliaroz, 2022-08-26:
@rufik 
Thank you for trying and time you spend for testing.

I have a  question to you
What official app do you use?

rufik, 2022-08-27:
@Iliaroz It's Huawei Body Fat Scale version  CH100_V1.1.11.120 from Play Store: https://play.google.com/store/apps/details?id=com.huawei.ch100
The scale is clearly Huawei AH-100 model.

I think getting online measurements is the crucial part here. It seems to work well, so maybe it should be merged into official app now? Offline measurements can be handled after that , I can help testing :)

Iliaroz, 2022-08-27:
@rufik 
While I test program on my own scale, there was the situation, when offline measurements was read by official app, but not by OpenScale. The reason is scale "think" that current measurement belong to user from officialapp. Check, please, do you have your offline measurements read by Body Fat Scale. You can try:
* change user profile in official app (add 5 kg to your weight)
* make online measurement from official app (with your new weight, take a 5 kg dumbell in your hands)
* make online measurement from OpenScale with own weight
* make offline measurement(s)
* make again online measurement from OpenScale , offline measurements should be read from scale (hopely :) )

Iliaroz, 2022-08-29:
@rufik 
Ok. The `----` mark for offline measures means that two or more app (from different devices) use the same scale for same user (~same weight of user +-2kg )
The `--` mark for offline measures means that there is no data for current user (weight) in scale.

When offline measurements show fat%, it means data is stored inside your scale for some user. 
Just to be sure: have you made online measurement from official app with new weight? And was the new weight for official app really +5kg to your own?

rufik, 2022-08-29:
So it looks offline measurements are being performed correctly on my scale.
Yes, I took online measurement in official app first, 84.8 kg, see attachment: 
[image]

Then I went to open scale app, all tried were between 81.0 and 82.0 kg. So 2.8kg difference at least.

martinriedel, 2025-12-26:
I've cracked the Honor CH100 (Huawei AH100) protocol. The key insight: measurement data is XOR-obfuscated only using the device MAC address, NOT device-specifically encrypted as previously believed.

  Weight algorithm found:
  weight_kg = (1457 - data[1]) / 10.0
  This is much simpler than the 5-bit algorithm proposed earlier. The data format:
  - Position 1: Weight (formula-encoded)
  - Position 2-3: Impedance (big-endian, Ω)
  - Position 20: Body fat % or skeletal muscle % (app-dependent)
  - Position 27: Body water %
  - Position 31: User ID

  Protocol: Two 16-byte frames (opcodes 0x0E + 0x8E) are merged and XOR'd with MAC. No AES encryption needed for measurements.

  Basic support is now working in openScale. The myth of "device-specific encryption" is busted! 🎉

Implementation available in PR #1245