oliexdev/openScale #381: Add Tenergy Vitalis support

foldsters, 2019-01-15:
Bluetooth logs from nRF Connect and the Tenergy Body app.

[Log 2019-01-14 20:21:10.txt](https://github.com/oliexdev/openScale/files/2757736/Log.2019-01-14.20.21.10.txt)
[image]

Other information not included in the app data is that I am 6'0".

I believe only the last couple of packets are significant. 

Let me know if you need anything else.

oliexdev, 2019-01-15:
thanks but we need the btsnoop_hci.log communication from the original app and your scale because we need also some information what we have to send to the scale. Read the [tutorial](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale).
Also we need to know the user information, read again above link step 1.3.

foldsters, 2019-01-15:
Ah, I'll take a look at that, thanks!