oliexdev/openScale #383: Support for Beurer BF 950

dror3go, 2019-01-15:
Before I go on to purchase the Beurer BF 950 - is it supported by the app? I couldn't find any reference to this model here.

Thanks!

oliexdev, 2019-01-15:
read [here](https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale) but it could be possible that they are using the same chip like BF700/710/800?!?

dror3go, 2019-01-16:
It's not in the Wiki, that's why I'm asking.
Would you advise me to purchase an older model instead, if I want to use it along with openScale? Or perhaps you'll help with debugging if needed? :smiley:

oliexdev, 2019-01-16:
Well, when it's not in the wiki with the title "supported scales in openScale" and it is not listed what would you guess if its supported or not!?
For the other question read the FAQ!

closed 2019-01-16 by oliexdev