oliexdev/openScale #386: Getting error

Dellu, 2019-01-20:
Build version: 1.8.4 
Current date: 2019-01-20 02:01:34 
Device: Sony G3311 
 
Stack trace:  
java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothGattCharacteristic android.bluetooth.BluetoothGattService.getCharacteristic(java.util.UUID)' on a null object reference
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:280)
	at com.health.openscale.core.bluetooth.BluetoothSenssun.sendUserData(BluetoothSenssun.java:90)
	at com.health.openscale.core.bluetooth.BluetoothSenssun.nextInitCmd(BluetoothSenssun.java:100)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStateStep(BluetoothCommunication.java:590)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.handleRequests(BluetoothCommunication.java:656)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.setBtMachineState(BluetoothCommunication.java:244)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$GattCallback$5.run(BluetoothCommunication.java:735)
	at android.os.Handler.handleCallback(Handler.java:836)
	at android.os.Handler.dispatchMessage(Handler.java:103)
	at android.os.Looper.loop(Looper.java:203)
	at android.app.ActivityThread.main(ActivityThread.java:6368)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1063)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:924)

Dellu, 2019-01-20:
The latest build gave me a different kind of error. This error doesn't occur anymore.

closed 2019-01-20 by Dellu

oliexdev, 2019-01-20:
@Dellu what kind of different error do you get now?