oliexdev/openScale #387: Weekly View Graph Bug

ericszimmermann, 2019-01-23:
Thank you for your great app.

I just recognized a somewhat courios behaviour in the Chart:
It happens, when you check the Option Week view (to see all Measurements) and you have a new Year's break. It could be similar to #147.
It Displays my Entry of 1.1.2019 on the far right of the Graph in Week 52 of the year 2019.
The first Point in year 2019 is said to be the 7.1.2019. (I measured every Day in between, because of medical reasons. This is also why I need the resolution of daily measurements in graph to see the trend.)
There seems to be a problem with calculating the Weeknumber in this graph setting.

Thank you for addressing this Issue
Eric Zimmermann

oliexdev, 2019-01-23:
could you export a csv dump, so that I can import it? If you are in the week view, you see the weeks not the days. Measurements in one week are then displayed as averages. If you click the average measurement you will get the last measurement in this average measurement.

ericszimmermann, 2019-01-23:
My Version of OpenScale is 1.8.4
In the attachmend i have put a sample CSV file with strictly decreasing
weight.
Pictures of the Weekly View toggle and the view in its current state.
There was a missunderstandig of the resolution on my side. Therefore I
would like to ask if it is possible to see all data in a graph? Because I
like to interpretate the Trend. I have lost 10kg in 3 weeks in my last
seizure. It always starts slowly and then the weight sinks with an
increasing rate. Therefore I like the regression funktion in your program.
In every case, you see the spike in the picture with the weekly view.

Thank you
EricZ

oliexdev, 2019-01-24:
There is no attachment, you can't attach anything via e-mail, please use GitHub. You could also test the [MPAndroidChart version](https://github.com/oliexdev/openScale/releases/tag/travis-dev_MPAndroidChart-build)

ericszimmermann, 2019-01-24:
Files under:
https://drive.google.com/open?id=1e77IoEPuZbJVf0vwml9PK5LvnJnuY5xA

oliexdev, 2019-01-24:
Sorry, I can't reproduce it with your data!? Maybe the problem is the display resolution of your smartphone? Do you have this effect if you rotate your screen orientation? Do you have the effect with the MPAndroidChart version too? Has the last measurement in 2018 really around 79kg?

ericszimmermann, 2019-01-24:
The data I delivered is in the Form:
3.1.2019, 70.0 kg
2.1.2019, 69.9 kg
1.1.2019, 69.8 kg
...
(Not my real dataset)

With the MPAndroidChart Version the year 2018 seems to be ok. But the year
has also this effect. Now its January and I have three Points on the Left
and one Point is on the far right. When clicking on it, it says to be the
Measurement of the Date 1.1.2019. So it looks like I have measured 3 Points
in January, 1 Point in Dezember (Future) and no Points in between.

Hope this helps
Eric Zimmermann

oliexdev, 2019-01-24:
Could you provide some screenshots again. It is better to explain this problem with images. 
But maybe they are a confusion about the week during the new year. In the week view in 2019 you have also the last week in 2018 because the new years eve was in the middle of the week! Only whole weeks are calculated in openScale.

ericszimmermann, 2019-01-24:
See new Picture
Screenshot_20290124...
In Google drive.

hirnsushi, 2019-01-27:
I'm having the same Problem:
[image]
Tried both the current dev version and the MPAndroidChart version, the problem persists.
Attached db should make this reproducible, at least it does on the different versions I tried. Had to zip em up, github wouldn't allow me to upload .csv or .db files for some reason.
[test-db.zip](https://github.com/oliexdev/openScale/files/2800475/test-db.zip)

hirnsushi, 2019-01-28:
The newest MPAndroidChart dev version fixed this, great work 👍

ericszimmermann, 2019-01-29:
I confirm that the graph now looks correct. Thanks for quickly fixing this
bug. Greetings Eric Zimmermann

oliexdev, 2019-01-29:
glad to hear :)

closed 2019-01-29 by oliexdev