oliexdev/openScale #388: screenshots missing in f-droid

BO41, 2019-01-23:
Currently, there are no screenshots of the app on the [f-droid page](https://f-droid.org/packages/com.health.openscale/).

I think you should add some (more = better), up-to-date ones.
People should see what the app looks like and what the features there are :)

https://f-droid.org/en/docs/All_About_Descriptions_Graphics_and_Screenshots/

oliexdev, 2019-01-24:
When the openScale version 2.0 is out, I will add some screenshots to f-droid because then the charts are replaced with MPAndroidChart.

oliexdev, 2019-02-22:
added a [merge request on GitLab](https://gitlab.com/fdroid/fdroiddata/merge_requests/4528)

oliexdev, 2019-03-04:
it was rejected so no screenshots for f-droid.

closed 2019-03-04 by oliexdev