oliexdev/openScale #389: [Feature] Scale graph to show goal by default

RichyHBM, 2019-01-25:
When on the graph view, the graph doesn't take into account your goal in the same way it does other measurements. 

Thus if you want to see your current progress with relation to your final goal you need to fiddle around with the graph to get them both in view.

This could also be a checkbox behind the burger menu on the chart view to be able to toggle it on or off

oliexdev, 2019-01-25:
try out the [MPAndroidChart](https://github.com/oliexdev/openScale/releases/tag/travis-dev_MPAndroidChart-build) version, if it works better in this version?

RichyHBM, 2019-01-25:
Just tried it and it doesn't seem like the goal line is there at all? There is only a dotted green line in the centre of the graph no matter what goal weight you have, more so you can't move the graph around anymore 😣

oliexdev, 2019-01-25:
yes the dotted green line is the goal line, see also the legend of the graph and no it seems only that the goal line is always in the centre because the y-axis dependency didn't follow the weight y-axis dependency but I have fixed this issue with commit https://github.com/oliexdev/openScale/commit/81356ba48d9ad79b4d617900a9533109a8d35bfb 
Please try again the latest MPAndroidChart build.
In the graph you can zoom in and out the graph but only in your selected month/year/week/day view. In the overview page you can scoll way back to the left freely.
For the view it is always automatically adjusted so that your data (goal line + measurement data) fit on the screen.

RichyHBM, 2019-01-25:
AHH yes, that's much better!

The only thing I would mention now is that it doesn't seem to log your initial weight (when you create a user) as an entry?

So I have to make a new entry for it to even register any data in the graphs

oliexdev, 2019-01-28:
yes the initial weight is only used for the "smart user assignment" if you have no measurement.

closed 2019-01-28 by oliexdev