oliexdev/openScale #390: Bluetooth connection to SBF75 fails with GATT_INSUF_AUTHORIZATION

berni-mt, 2019-01-26:
Since the latest update (v1.9 (37)) via play store, I can't connect to my SBF75 anymore. The app is repeatedly trying to connect and failing again until the error with status 8 (GATT_INSUF_AUTHORIZATION) is shown. 

I've reset the scale and also deleted cache of openScale but nothing helped so far. Please have a look into the attached logfile.
[openScale_2019-01-26_11-53.txt](https://github.com/oliexdev/openScale/files/2798913/openScale_2019-01-26_11-53.txt)

My smartphone is a Sony XZ1C (G8441) running on Android 9 (Build 47.2.A.4.45).

berni-mt, 2019-01-26:
Well, the dev version worked quite well 👍

oliexdev, 2019-01-29:
great should be fixed in the release version v1.9.1

closed 2019-01-29 by oliexdev