oliexdev/openScale #392: MiScale data from all users

rchovan, 2019-01-29:
**Describe the bug**
When I have connected to Mi Scale, app loads data for all users. 
openScale version 1.8.4

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): not tried

**Expected behavior**
Load data for current user

**Additional context**
screenshot 
[image]

**Debug log**
not enabled

oliexdev, 2019-01-29:
Did you enable the "smart user assignment"? Did you create different users in openScale with different initial weights?

rchovan, 2019-01-29:
Tested again with this settings
[image]

First measurement is correct. Old data are read after next measurement. Attaching debug log.
```
2019-01-29 17:51:31.341 Debug [2] AboutPreferences: Debug log enabled, openScale v1.8.4 (32), SDK 26, samsung SM-G950F
2019-01-29 17:52:00.524 Debug [2] OpenScale: Select user 3
2019-01-29 17:52:53.115 Debug [2] OpenScale: Trying to connect to bluetooth device [E8:7C:5E:94:C6:83] (MIBCS)
2019-01-29 17:52:53.132 Debug [2] BluetoothCommunication: BT is enabled, state=12, scan mode=21, is not discovering
2019-01-29 17:52:53.141 Debug [2] BluetoothCommunication: Connected GATT device: vívoactive [D7:01:62:44:B8:DB]
2019-01-29 17:52:53.146 Debug [2] BluetoothCommunication: Connected GATT_SERVER device: vívoactive [D7:01:62:44:B8:DB]
2019-01-29 17:52:53.151 Debug [2] BluetoothCommunication: Starting LE scan for device [E8:7C:5E:94:C6:83]
2019-01-29 17:52:53.336 Debug [2] BluetoothCommunication: Found LE device MIBCS [E8:7C:5E:94:C6:83]
2019-01-29 17:52:53.337 Info [2] BluetoothCommunication: Connecting to [E8:7C:5E:94:C6:83] (driver: Xiaomi Mi Scale v2)
2019-01-29 17:52:53.895 Debug [171710] BluetoothCommunication$GattCallback: onConnectionStateChange: status=0, newState=2
2019-01-29 17:52:53.901 Debug [2] BluetoothCommunication: Stopping LE scan
2019-01-29 17:52:53.947 Debug [2] MainActivity: Bluetooth connection successful established
2019-01-29 17:52:54.942 Debug [171710] BluetoothCommunication$GattCallback: onServicesDiscovered: status=0 (5 services)
2019-01-29 17:52:55.954 Debug [2] BluetoothCommunication: INIT STATE: 0
2019-01-29 17:52:55.963 Debug [2] BluetoothCommunication: Write characteristic 00001542-0000-3512-2118-0009af100700: 06 04 00 00 (type: 2; queue: 0 0)
2019-01-29 17:52:56.095 Debug [2] BluetoothCommunication: INIT STATE: 1
2019-01-29 17:52:56.104 Debug [2] BluetoothCommunication: Write characteristic 0x2a2b "current time": E3 07 01 1D 11 34 38 03 00 00 (type: 2; queue: 0 0)
2019-01-29 17:52:56.288 Debug [2] BluetoothCommunication: INIT STATE: 2
2019-01-29 17:52:56.291 Debug [2] BluetoothCommunication: Set notification on for 00002a2f-0000-3512-2118-0009af100700
2019-01-29 17:52:56.297 Debug [2] BluetoothCommunication: Write descriptor 0x2902 "client characteristic configuration": 01 00 (queue: 0 0)
2019-01-29 17:52:56.433 Debug [2] BluetoothCommunication: INIT STATE: 3
2019-01-29 17:52:56.438 Debug [2] BluetoothCommunication: CMD STATE: 0
2019-01-29 17:52:56.448 Debug [2] BluetoothCommunication: Write characteristic 00002a2f-0000-3512-2118-0009af100700: 01 FF FF AF AD (type: 2; queue: 0 0)
2019-01-29 17:52:56.671 Debug [171734] BluetoothCommunication$GattCallback: onCharacteristicChanged 00002a2f-0000-3512-2118-0009af100700: 01 00 00 FF FF AF AD
2019-01-29 17:52:56.681 Debug [2] BluetoothMiScale2: DataChange hex data: 01 00 00 FF FF AF AD
2019-01-29 17:52:56.726 Debug [2] BluetoothCommunication: CMD STATE: 1
2019-01-29 17:52:56.729 Debug [2] BluetoothCommunication: Set notification off for 00002a2f-0000-3512-2118-0009af100700
2019-01-29 17:52:56.735 Debug [2] BluetoothCommunication: Write descriptor 0x2902 "client characteristic configuration": 00 00 (queue: 0 0)
2019-01-29 17:52:56.877 Debug [2] BluetoothCommunication: CMD STATE: 2
2019-01-29 17:52:56.882 Debug [2] BluetoothCommunication: Set notification on for 00002a2f-0000-3512-2118-0009af100700
2019-01-29 17:52:56.892 Debug [2] BluetoothCommunication: Write descriptor 0x2902 "client characteristic configuration": 01 00 (queue: 0 0)
2019-01-29 17:52:57.070 Debug [2] BluetoothCommunication: CMD STATE: 3
2019-01-29 17:52:57.077 Debug [2] BluetoothCommunication: Write characteristic 00002a2f-0000-3512-2118-0009af100700: 02 (type: 2; queue: 0 0)
2019-01-29 17:52:57.157 Debug [171734] BluetoothCommunication$GattCallback: onCharacteristicChanged 00002a2f-0000-3512-2118-0009af100700: 03
2019-01-29 17:52:57.164 Debug [2] BluetoothMiScale2: DataChange hex data: 03
2019-01-29 17:52:57.169 Debug [2] BluetoothCommunication: Request pending (queue 0 0)
2019-01-29 17:52:57.236 Debug [2] BluetoothCommunication: CLEANUP STATE: 0
2019-01-29 17:52:57.239 Debug [2] BluetoothCommunication: Write characteristic 00002a2f-0000-3512-2118-0009af100700: 03 (type: 2; queue: 0 0)
2019-01-29 17:52:57.407 Debug [2] BluetoothCommunication: CLEANUP STATE: 1
2019-01-29 17:52:57.414 Debug [2] BluetoothCommunication: Write characteristic 00002a2f-0000-3512-2118-0009af100700: 04 FF FF AF AD (type: 2; queue: 0 0)
2019-01-29 17:52:57.654 Debug [2] BluetoothCommunication: CLEANUP STATE: 2
2019-01-29 17:52:57.660 Debug [2] BluetoothCommunication: Set notification on for 0x2a9c "body composition measurement"
2019-01-29 17:52:57.672 Debug [2] BluetoothCommunication: Write descriptor 0x2902 "client characteristic configuration": 01 00 (queue: 0 0)
2019-01-29 17:52:57.844 Debug [2] BluetoothCommunication: CLEANUP STATE: 3
2019-01-29 17:53:00.222 Debug [171734] BluetoothCommunication$GattCallback: onCharacteristicChanged 0x2a9c "body composition measurement": 02 04 E3 07 01 1D 11 35 00 00 00 54 06
2019-01-29 17:53:00.225 Debug [2] BluetoothMiScale2: DataChange hex data: 02 04 E3 07 01 1D 11 35 00 00 00 54 06
2019-01-29 17:53:00.418 Debug [171734] BluetoothCommunication$GattCallback: onCharacteristicChanged 0x2a9c "body composition measurement": 02 04 E3 07 01 1D 11 35 00 00 00 40 15
2019-01-29 17:53:00.422 Debug [2] BluetoothMiScale2: DataChange hex data: 02 04 E3 07 01 1D 11 35 00 00 00 40 15
[... cut ...]
2019-01-29 17:53:32.300 Debug [171734] BluetoothCommunication$GattCallback: onCharacteristicChanged 0x2a9c "body composition measurement": 02 A6 E3 07 01 1D 11 35 18 E5 01 4C 40
2019-01-29 17:53:32.304 Debug [2] BluetoothMiScale2: DataChange hex data: 02 A6 E3 07 01 1D 11 35 18 E5 01 4C 40
2019-01-29 17:53:55.077 Debug [2] OpenScale: Disconnecting from bluetooth device
2019-01-29 17:53:55.086 Info [2] BluetoothCommunication: Disconnecting (with cleanup)
2019-01-29 17:53:55.301 Info [2] BluetoothCommunication: Disconnecting
```

closed 2019-01-30 by oliexdev

rchovan, 2019-01-30:
Hello @oliexdev,
could you please provide answer ? What I'm receiving all of the data ? Is something wrong with scale, application or settings ? 

Thank You .

oliexdev, 2019-01-30:
Well, you didn't ask anything in your previous post. The mi scale doesn't delete any data on your scale. Duplicate data (that are already imported) are ignored by openScale.

rchovan, 2019-01-30:
I'm confused. This scale is used by me and my wife, and our kids without application. She uses MiFit application I use openScale. . From my point of view, they are not ignored by openScale (See image from post 1. ) I have tried it again and attached debug log to post 3. I did only two measurement and openScale show measurement from past, which are not my measurement. So do I miss something ? Or it is normal behaviour ?

oliexdev, 2019-01-30:
Yes it is a normal behaviour. They are ignored see your log file. 
The measurements are assign to the user with the nearest weight value, if you have enable the smart user assignment.
Clean you openScale measuremetns. Then you have to set all you family member in openScale with their initial weight. Then sync to the scale with smart user assignment on. If you measurements are overlapped between some users then it is a problem in the mi scale app and in openScale.

rchovan, 2019-01-30:
This is exactly what I did yesterday. I have deleted my user in openScale application, and created again. But I have received measurement which do not belongs to me. As you can see on this image, it shows data older tan 1 day, which from my point of view is incorrect, because I started as new user, and measure my self only 2 times and only in yesterday. 

[image]

And when I switch to chart view, and switch year 2018 I see remaining measurements. 

[image]

How to filter data which do not belongs to me ?

oliexdev, 2019-01-30:
as I already wrote the scale never delete any data! How many users did you create?