oliexdev/openScale #393: Add milestone

RichyHBM, 2019-01-29:
Hello, it would be useful to have a milestone in GitHub to view what issues are holding up the next release. 

Are there many issues holding back v2.0?

Also it seems on Play Store the latest version is 1.8.4 where as the latest stable version on both GitHub and f-droid is 1.9

oliexdev, 2019-01-29:
The latest release is version [v1.9.1](https://github.com/oliexdev/openScale/releases)
On GooglePlay I released it step by step (currently 30%) to see if this version have huge bugs.
On f-droid we have to wait until the f-droid bot scan again the openScale tags (it can take up to one week) if it on f-droid available. 

What did you expect from v2.0?

In general I don't have a milestone plan. I will release the next version when we have major changes or major bugs. On v2.0 I will change the chart library with MPAndroidChart library, see #371.

RichyHBM, 2019-01-29:
It would just be to get a better idea of how long v2 will take.

I'm very much looking forward to it but I don't like installing apps out of f-droid or play store

closed 2019-01-30 by oliexdev