oliexdev/openScale #397: Measurements not saving anymore on Sanitas SBF70

jobtijhuis, 2019-02-11:
When using openScale with the SBF70 the measurements are not saved anymore (it used to work before). It stopped working somewhere around the 22th of January.

The measurement does pop up and it is correct and I would expect it to be saved, but it is not. (With popping up I mean the gray bubble at the bottom of the screen showing the current weight on the scale)

If there are any logs I can provide I am happy to do so (I am just not entirely familiar with Android logging).

oliexdev, 2019-02-11:
install the latest [dev build](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) and turn on the openScale log under settings->about then try to connect to your scale again. Finally attach this log here.

erijo, 2019-02-11:
Perhaps related to #398.

jobtijhuis, 2019-02-14:
It was indeed related to #398 and I can confirm the latest developments build fixes the problem.

closed 2019-02-14 by jobtijhuis

oliexdev, 2019-02-16:
@all please test the version linked in the issue #400