oliexdev/openScale #398: SBF-70 not clearing history on the sclae anymore

lenucksi, 2019-02-11:
**Describe the bug**
Using version 1.9.3 and a SBF-70 scale I was able to retrieve measurements (better now with the external bluetooth library) however, the entries in the scale's memory were not cleared leaving the scale in showing "FULL" only not saving any further measurements. 

The toast message on new measurements appears though. The scale show that it connects fine, the apps reports a successful connection, too.

**Expected behavior**
Measurements should be fetched and removed from the scale after a successful fetch.
[openScale_2019-02-11_10-17.txt](https://github.com/oliexdev/openScale/files/2850896/openScale_2019-02-11_10-17.txt)
[openScale_2019-02-04_08-03.txt](https://github.com/oliexdev/openScale/files/2850897/openScale_2019-02-04_08-03.txt)

**Debug log**
Added logs from two attempts to fetch measurements should be from 1.9.3. and 1.9.1.

oliexdev, 2019-02-11:
Sorry, I don't know why it is not working. I didn't change the logical flow of this scale!? I can't really help because I don't own this scale. We have to wait if somebody looks deeper into this problem. In the meantime maybe a reset of the scale will help?

erijo, 2019-02-11:
I think the problem can be seen here:
```2019-02-11 10:18:06.412 Debug [2] BluetoothCommunication: Write characteristic 0xffe1: E7 F1 42 3C 3B
2019-02-11 10:18:06.425 Debug [2] BluetoothCommunication: onCharacteristicChanged 0xffe1: E7 42 3C 3C FE 01 89 01 08 07 D9 0A FD 00 FA
2019-02-11 10:18:06.476 Debug [4897] RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(155894492)
2019-02-11 10:18:06.478 Debug [4896] RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(80400814)
2019-02-11 10:18:06.481 Debug [4894] RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(80400814)
2019-02-11 10:18:06.551 Debug [4882] RxBle#BluetoothGatt$1: onCharacteristicWrite characteristic=0000ffe1-0000-1000-8000-00805f9b34fb status=0
2019-02-11 10:18:06.554 Debug [2] BluetoothCommunication: Write characteristic 0xffe1: E7 43 00 00 00 00 00 00 00 65
2019-02-11 10:18:06.555 Debug [4894] RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(80400814) in 74 ms
2019-02-11 10:18:06.555 Debug [4894] RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(155894492)
2019-02-11 10:18:06.655 Debug [4902] RxBle#BluetoothGatt$1: onCharacteristicWrite characteristic=0000ffe1-0000-1000-8000-00805f9b34fb status=0
2019-02-11 10:18:06.660 Debug [2] BluetoothCommunication: Write characteristic 0xffe1: E7 F1 42 3C 3C
2019-02-11 10:18:06.661 Debug [4894] RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(155894492) in 103 ms
```

After receiving the last measurement (E7 42 3C 3C...) openScale is supposed to send an ack (E7 F1 42 3C 3C), but that ack is sent after the command to delete all measurements (E7 43 00 00 00 00 00 00 00 65).

Judging from the logs from RxBle it seems like the write operations are queued in the right order, but then executed in the wrong order.

oliexdev, 2019-02-11:
@erijo could be possible, regarding to the code below
```
    private void processSavedMeasurement(byte[] data) {
        int count = data[2] & 0xFF;
        int current = data[3] & 0xFF;

        processMeasurementData(data, 4, current % 2 == 1);
        sendAck(data);

        if (current == count) {
            sendCommand(CMD_DELETE_SAVED_MEASUREMENTS, encodeUserId(currentRemoteUser));
        }
    }
```

the `sendAck()` and `sendCommand()` are not in the BT state machine. Could it possible to do the `sendCommand()` in the last machine cmd step? or is a notify send from the scale that the last measurement is received?

erijo, 2019-02-11:
But I don't see why that should be changed. Why doesn't RxBle execute the operations in the same order they are queue?

oliexdev, 2019-02-11:
RxBle is stateless and asynchron if two write command are executed parallel one could be faster than the other one. You could sequential it with the RxJava techniques but that we can't use because then the BluetoothCommunication wouldn't be abstract anymore.

erijo, 2019-02-11:
@lenucksi: please try with the latest dev version once it has been built and attach the log here.

closed 2019-02-11 by erijo

oliexdev, 2019-02-16:
@all please test the version linked in the issue #400

commit 7670e56a (2019-02-11): Do delete saved measurements as a separate step

Hopefully this fixes #398, but note that the code is untested.