oliexdev/openScale #400: Bluetooth state machine 2.0

oliexdev, 2019-02-16:
I refactored the internal Bluetooth state machine to be more flexible in the programming.

Please download and install the [**openScale Bluetooth state machine 2.0 apk**](https://github.com/oliexdev/openScale/releases/tag/travis-dev_BluetoothStateMachine2.0-build) and test this version with your scales.

Please report back if it works or not. If something doesn't work please attach the openScale log file (enable it under settings->about)

Thanks for your help in advance.

jobtijhuis, 2019-02-17:
I have tested it and I have the same issue as I mentioned in #397. There is communication with the scale and it does show the current weight on the scale, but the measurement is not added to the graph.

I have attached the log.
[openScale_2019-02-17_11-11.txt](https://github.com/oliexdev/openScale/files/2872907/openScale_2019-02-17_11-11.txt)

lenucksi, 2019-02-17:
I've also tested it with the SBF-75 and found it work worse than the old revision - there even appeared to be connection issues. I'll give the next build a try again.

oliexdev, 2019-02-20:
Thanks I fixed the problem please try again the latest [**openScale Bluetooth state machine 2.0 apk**](https://github.com/oliexdev/openScale/releases/tag/travis-dev_BluetoothStateMachine2.0-build)

oliexdev, 2019-02-21:
merged with PR https://github.com/oliexdev/openScale/pull/404
Test the latest dev version [here](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build).

closed 2019-02-21 by oliexdev