oliexdev/openScale #401: support for Picooc Mini scale

artsight, 2019-02-16:


oliexdev, 2019-02-16:
Please provide more information regarding to https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale

artsight, 2019-02-17:
> Please provide more information regarding to https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale

Added both files, sorry, was not able to do that from mobile.

oliexdev, 2019-02-17:
the true values are missing for both files

artsight, 2019-02-18:
> the true values

sorry, what you mean by that? I see the logfiles are not empty, what is missing?

unfortunately I don't have wireshark to look into bt snoop deeply

artsight, 2019-02-18:
Here is screenshot from native app with all values. Tried to find them in the btsnoop log but ...
[image]

oliexdev, 2019-02-18:
is this the screenshot from the above log file? Please do it again with different weights described in https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale (see step 1.6)

artsight, 2019-02-19:
yes, the screenshot above is based on the logfile attached previously.

I made 4 other measurements with different weights, please see the attached zip.
[picooc mini logs.zip](https://github.com/oliexdev/openScale/files/2879331/picooc.mini.logs.zip)

oliexdev, 2019-02-20:
which screenshots belongs to which log?

artsight, 2019-02-22:
> which screenshots belongs to which log?
Sorry, didn't mention that:
2,3,4.jpg - btsnoop_hci.20190219092631
5.jpg - btsnoop_hci.20190219092904

2m, 2021-07-29:
I am also interested in support for this scale. Is there another BT capture required for this to be completed?

plant7403, 2024-12-25:
Anyone? 😁 please?
As it seems almost all of the models of picooc have very similar ways they synchronize
So it's possible to add like a lot of scales at once with this one