oliexdev/openScale #402: Pivotal Living Smart Scale Support

aftereffectsmagic, 2019-02-18:
Hi there, 

Looking for openScale support for my Pivotal Living Smart Scale, as the platform went belly up several years ago and the app no longer works. Here are the openScale logs, along with the btsnoop log. 

[openScale_2019-02-17_22-06.txt](https://github.com/oliexdev/openScale/files/2874256/openScale_2019-02-17_22-06.txt)
[btsnoop_hci_motog.log](https://github.com/oliexdev/openScale/files/2874257/btsnoop_hci_motog.log)

Weight is 95.0276 kg (which is E8004 in hex, and apparently something like 00 04 80 0E in little endian (float)) but I'm not sure my calculations are correct as I wasn't able to find 00 04 80 0E in wireshark when looking at the btsnoop log. My first time doing that so any pointers would be great. Additionally, everything in that log is from either controller() to host() or host() to controller() (not remote() and localhost() as stated in the Reverse Engineer instructions). 

Glad to jump through requisite hoops to get this working :) 

Thanks and cheers!

oliexdev, 2019-02-18:
Where do you have the weight 95.0276kg from? It is displayed from by the original app? Three decimals places after comma is really unusual!?

Please do the steps exactly described on https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale (You missed for example step 1.3 and you did the whole step 2 wrong)

> Additionally, everything in the btsnoop log is from either controller() to host() or host() to controller() (not remote() and localhost() as stated in the Reverse Engineer instructions). Does that matter?

No, it doesn't matter sometimes it is called remote / host  or the device name is written depends on the wireshark version.

Your bytes 
`04 0e 05 01 0f 20 00 80`
are just an ACK command.

jhoff909, 2020-05-06:
How can I help getting support for Pivotal Living scale?  I have one but the original app is not available and I never had it.