oliexdev/openScale #405: openscale dev error occur

hiro-bkk, 2019-02-23:
I would like to post bt log for issue my scale support.
But cannot scan with openscale dev ver.
please confirm and improve.

**Describe the bug**
Openscale dev version error occur when press bluetooth scale button.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '"Setting'
2. Click on 'Bluetooth'
3. And then click on  'Your Bletooth scale'
4. See error

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): v2.0 dev

**Expected behavior**
scan bluetooth scale

**Additional context**
Android 4.4.4

**Debug log**
Build version: 2.0-dev_31db8188_2019-02-22 
Current date: 2019-02-23 15:51:41 
Device: HUAWEI d-01G 
 
Stack trace:  
java.lang.SecurityException: "gps" location provider requires ACCESS_FINE_LOCATION permission.
	at android.os.Parcel.readException(Parcel.java:1468)
	at android.os.Parcel.readException(Parcel.java:1422)
	at android.location.ILocationManager$Stub$Proxy.isProviderEnabled(ILocationManager.java:933)
	at android.location.LocationManager.isProviderEnabled(LocationManager.java:1128)
	at com.health.openscale.gui.utils.PermissionHelper.requestLocationServicePermission(PermissionHelper.java:90)
	at com.health.openscale.gui.preferences.BluetoothPreferences$5.onPreferenceClick(BluetoothPreferences.java:235)
	at android.preference.Preference.performClick(Preference.java:954)
	at android.preference.PreferenceScreen.onItemClick(PreferenceScreen.java:228)
	at android.widget.AdapterView.performItemClick(AdapterView.java:302)
	at android.widget.AbsListView.performItemClick(AbsListView.java:1113)
	at android.widget.AbsListView$PerformClick.run(AbsListView.java:2911)
	at android.widget.AbsListView$3.run(AbsListView.java:3650)
	at android.os.Handler.handleCallback(Handler.java:733)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:136)
	at android.app.ActivityThread.main(ActivityThread.java:5274)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:515)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:851)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:667)
	at dalvik.system.NativeStart.main(Native Method)

oliexdev, 2019-02-23:
please try out the latest dev build.

hiro-bkk, 2019-02-24:
Thank you very much.
I cofirmd to resolve.

hiro-bkk, 2019-02-24:
Error occur again.

Scand my scale and then found.
Select my scale but error occurred.

Build version: 2.0-dev_2fd331ad_2019-02-23 
Current date: 2019-02-24 08:44:14 
Device: HUAWEI d-01G 
 
Stack trace:  
java.lang.NullPointerException
	at com.health.openscale.core.bluetooth.BluetoothDebug.onNextStep(BluetoothDebug.java:176)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:569)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:571)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.access$600(BluetoothCommunication.java:53)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.run(BluetoothCommunication.java:511)
	at android.os.Handler.handleCallback(Handler.java:733)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:136)
	at android.app.ActivityThread.main(ActivityThread.java:5274)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:515)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:851)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:667)
	at dalvik.system.NativeStart.main(Native Method)

oliexdev, 2019-02-24:
thanks should be fixed with https://github.com/oliexdev/openScale/commit/39d41ce0ec2199d56936e41a81f02c54f409761e

hiro-bkk, 2019-02-24:
Resolved with 49823d3.
Thank you.

closed 2019-02-24 by hiro-bkk