oliexdev/openScale #407: Version 2 mixes up different users

lowtower, 2019-02-24:
In version 2.0 in the table tab, the results of different users are being mixed up.
I have two users (ca. 50kg and 70kg).
In the last measurement, I gained twenty kg, because my weight ist being compared to the other users weight.

oliexdev, 2019-02-24:
does your two users weights are overlapping now or near each other?

lowtower, 2019-02-24:
Uhm, misunderstanding, I suppose!
User 1 still 50kg.
User 2 still 70kg.
But in Your Software both user's measurements appear in the same table.
Measurement 1, User 1: 50kg
Measurement 2, User 1: 51kg -> +1kg
Measurement 3, User 2: 71kg -> +20kg
Measurement 4, User 1: 50kg -> -21kg
In the software versions prior to 2.0, only values of one single user were shown

oliexdev, 2019-02-24:
hmm can't reproduce it. Do you sync it with a scale? What is shown on the scale? Did you enabled the smart user assignment? What do you mean with +20kg / -21kg? is these are the new measurement?
Which user did you selected when you add a measurement? Please give a step by step example, so I can reproduce it.

lowtower, 2019-02-25:
Okay, one by one:

I have a smartphone with android 7.0.
I have a "Sanitas SBF70" scale.
Two users are registered on the scale, no major weight changes of real persons:
- user1: ~50kg
- user2: ~70kg
Both users are registered in the app, only user2 synchronizes his weigth with the scale
In settings, intelligent user detection is **activated**

**App version 1.9.3** is installed:
- Tab "Home":
  - **user2** is selected in dropdown
  - tab shows latest measurements of user2
- Tab "Diagram":
  - tab shows changes in measured values over time as a diagram
  - only values of user2 are shown
- Tab "Table":
  - tab shows table of measured values
  - only values of user2 are shown

App is upgraded to **App version 2.0**:
- Tab "Home":
  - **user2** is selected in dropdown
  - tab shows latest measurements of user2 **and** user1 mixed up
  - all values are mixed up
  - e.g. weight changes between user1 and user2 at about 20kg plus and minus, depending on which weight was measured the latest
- Tab "Diagram":
  - tab shows changes in measured values over time as a diagram
  - measured values of both users, user1 **and** user2 are shown
- Tab "Table":
  - tab shows table of measured values
  - measured values of both users, user1 **and** user2 are shown
- **user1** is selected in dropdown
  - only measured values of user1 are shown, as expected

As far as I understand, the Sanitas SBF70 has a memory of ten or twenty measurements.
I expect the app to show only the values of the selected user on all tabs.
Obviously some measurements have been detected to be of user2, while they should have been of user1!

lowtower, 2019-02-25:
I deleted the wrong assigned measurements and for now all is correctly displayed.
Only some measurements have been assigned to the wrong user
Is the assignment being done in the app or on the scale?
BTW: is it possible to re-assign a measurement to a different user in the app?

I would like to keep this open and make some further measurements and check the assignment.

lowtower, 2019-02-28:
Hello @oliexdev,

I was able to reproduce the issue, with following steps:
- user 1 uses the scale, no synchronisation with openscale
- user 1 uses the scale, no synchronisation with openscale
- user 2 uses the scale, synchronises with openscale
- openscale retrieves three measurements
- in the app:
  - user 1 has two new measurements assigned -> correct
  - user 2 has three new measurements assigned -> incorrect, should be only one
  - if I delete one false measurement for user 2, the same measurement is not deleted for user 1

It seems that the user assignment in the app is not working correctly since version 2.0.

btw.:
user 1 does **not** wait for user assignment in the scale.
Only weight is captured, but not the other values, such as "body water", muscle", etc.
Don't know if it makes a difference for user assignment in the app.

IronJunkie, 2019-03-10:
My wife has same bug, on Samsung S8 running Android 8 Oreo, after update to Feb 24th release from Play store.

Only difference to OP is that all users on this scale sync automatically to open scale app via Bluetooth. Also the graph plots incorrectly ie plots wrong  persons data, so error not just in table. Sorry didn't have log switched on.

(also she has the problem of multiple measurements eg 20 being taken & showing in table, as per issue "Senssun scale" #417 )

openScale v2.0.1 (42), SDK 26, samsung SM-G950F
scales: SENSSUN

oliexdev, 2019-03-10:
@IronJunkie senssun scale is not supporting multiple users!? You have to switch the users in openScale before you using your scale. The problem in issue #417 seems to be fixed with #427 

@lowtower I reproduce exactly your steps and it works as expected. Every user has their correct weight assigned.

Some comments of your scale which I think you misunderstand a bit.

>  Is the assignment being done in the app or on the scale?

it is done on the scale! You have firstly registered the user on the scale with the first weight.

> is it possible to re-assign a measurement to a different user in the app?

No currently that is not supported, might be in the future.

> user 1 does not wait for user assignment in the scale.
> Only weight is captured, but not the other values, such as "body water", muscle", etc.
> Don't know if it makes a difference for user assignment in the app.

That should be the problem. If no user was assigned by the scale, the weight will be assigned to the current selected openScale user.

> if I delete one false measurement for user 2, the same measurement is not deleted for user 1

Yeah sure hopefully the other measurement is not deleted. That would be totally wrong because it is a measurement of another user!

closed 2019-03-10 by oliexdev

lowtower, 2019-03-10:
> it is done on the scale! You have firstly registered the user on the scale with the first weight.
Both users are registered on the scale
> Did you enabled the smart user assignment?
What's the sense of this when the user assignment is done on the scale only.
In the versions prior to 2.0 it worked as expected.

lowtower, 2019-03-10:
> Yeah sure hopefully the other measurement is not deleted. That would be totally wrong because it is a measurement of another user

Yes indeed!
But it is assigned to both users in the App!
One single measurement exists twice.
That's exactly the problem!

oliexdev, 2019-03-10:
> What's the sense of this when the user assignment is done on the scale only.

Actually, with your scale this option doesn't have any effect.