oliexdev/openScale #408: Improvments for V2.0

ericszimmermann, 2019-02-24:
# Improvments for V2.0

I am realy happy the see the new charts merged into the stable release.
Thank you for your effort and this great app.
But there are some Improvements which could be made:

## Front Graph
- In Dark mode: Y-Achsen  are not grey, but black. Therefore badly readble
- Regression line is not working: Flat line on Zero. This causes the scalerange of the  Overview not to be around the measured values, but starting zero. Thus making the weight (only value enabled) also almost a flat line.
- Indicators (-0.7kg) are not colored.

## Diagram
- also the first two point from above
- at startup only the weight is selected on  the round measuremend buttons, whereas in the graph all the selected values are shown.

## Table
- In Dark mode: Symbols are black, therefore not good readable. Supposing to make grey or colored circles behind it. (For not having to make a second white set of symbols)

oliexdev, 2019-02-24:
> In Dark mode: Y-Axes are not grey, but black. Therefore badly readble

fixed

> Regression line is not working: Flat line on Zero. This causes the scale-range of the  overview not to be around the measured values, but starting by zero. Thus making the weight (only value enabled) also almost a flat line.

I had to disable the regression line. Might be replace in the future with an sliding average line.

> Indicators (-0.7kg) are not colored.

Do you mean in the graph or in the detail view? in the detail view nothing its coloured. In the graph it works as expected.

> at startup only the weight is selected on the round measuremend buttons, whereas in the graph all the selected values are shown.

That's a correct behaviour. You can select your showed measurements individual for each graph. Turn on the measurement bar.

> In Dark mode: Symbols are black, therefore not good readable. Supposing to make grey or colored circles behind it. (For not having to make a second white set of symbols)

should be fixed

ericszimmermann, 2019-02-26:
Sear oliexdev
Now I have a Computer at my Hand, so I can upload some screenshots to show what I ment:

>     * Indicators (-0.7kg) are not colored.
I'm not sure, but I thought on the previous version (old chart) these were coloured. Would be nice if it could be here too.
I also would find it quite nice, if you could put in the date of the last measurement, espacialy because the points in the graph are not strictly on the grid. (Date in the Graph is displayed at Midday 12 am.)
[image]

> at startup only the weight is selected on the round measuremend buttons, whereas in the graph all the selected values are shown.
As soon as I press one of the Measures, they get colored (enabled/disabled) correctly.
[image]

I hope my reclamations aren't too annoying and hope, that i'm helping.

oliexdev, 2019-02-27:
your points are already fixed in the dev version but the coloured indicator arrow are only colored in the graph itself.

Don't worry you had valid points and I am thankful for your help.

closed 2019-02-27 by oliexdev