oliexdev/openScale #410: Support request for BL-26L01

hiro-bkk, 2019-02-24:
Hello.
Could you please support for my scale BL-26L01.
This scale support only for OKOK. International. 

[openScale_2019-02-24_20-10.txt](https://github.com/oliexdev/openScale/files/2897932/openScale_2019-02-24_20-10.txt)
[openScale_2019-02-24_20-13.txt](https://github.com/oliexdev/openScale/files/2897936/openScale_2019-02-24_20-13.txt)
[01btsnoop_hci.log](https://github.com/oliexdev/openScale/files/2897938/01btsnoop_hci.log)
[02btsnoop_hci.log](https://github.com/oliexdev/openScale/files/2897939/02btsnoop_hci.log)
[03btsnoop_hci.log](https://github.com/oliexdev/openScale/files/2897941/03btsnoop_hci.log)

Thank you.

oliexdev, 2019-02-24:
please do **all** steps in the wiki probably.

hiro-bkk, 2019-03-02:
I found only weight.
01btsnoop_hci.log  weight 68.00kg -> 1A 90
"98","7.355170","controller","host","HCI_EVT","46","Rcvd LE Meta (LE Advertising Report)"

02btsnoop_hci.log  weight 71.30kg => 1B DA
"140","14.101665","controller","host","HCI_EVT","46","Rcvd LE Meta (LE Advertising Report)"

03btsnoop_hci.log  weight 68.65kg  => 1A D1
"77","6.188214","controller","host","HCI_EVT","46","Rcvd LE Meta (LE Advertising Report)"

inzanity, 2021-03-04:
Ok, this seems to send out slightly different advertisements than mine, there's actually a product id (which we probably don't care about: 0x41 0xaf 0x2f 0x81) and sending 0x04 for in-progress results, and 0x05 for final results, and weight is sent as *100. So it seems that my PR #706 needs some updating to use the bit 2 for granularity.  I wonder if bit 1 has any meaning...