oliexdev/openScale #411: Immediate crash

loetsinn, 2019-02-24:
OpenScale 2.0 crashes immediately after opening the app. Previous veraion was running fine. This is the error code i got:

```
Build version: 2.0 
Build date: 1980-01-01 00:00:00 
Current date: 2019-02-25 00:05:14 
Device: Samsung SM-T320 
 
Stack trace:  
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.health.openscale/com.health.openscale.gui.MainActivity}: android.view.InflateException: Binary XML file line #126: Binary XML file line #126: Error inflating class <unknown>
at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2432)
at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2492)
at android.app.ActivityThread.-wrap11(ActivityThread.java)
at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1360)
at android.os.Handler.dispatchMessage(Handler.java:102)
at android.os.Looper.loop(Looper.java:148)
at android.app.ActivityThread.main(ActivityThread.java:5439)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
Caused by: android.view.InflateException: Binary XML file line #126: Binary XML file line #126: Error inflating class <unknown>
at android.view.LayoutInflater.inflate(LayoutInflater.java:539)
at android.view.LayoutInflater.inflate(LayoutInflater.java:423)
at com.health.openscale.gui.fragments.GraphFragment.onCreateView(GraphFragment.java:111)
at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2440)
at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:885)
at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1229)
at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:1295)
at androidx.fragment.app.FragmentManagerImpl.dispatchStateChange(FragmentManagerImpl.java:2605)
at androidx.fragment.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManagerImpl.java:2565)
at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:245)
at androidx.fragment.app.FragmentActivity.onStart(FragmentActivity.java:525)
at androidx.appcompat.app.AppCompatActivity.onStart(AppCompatActivity.java:179)
at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1238)
at android.app.Activity.performStart(Activity.java:6268)
at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2395)
... 9 more
Caused by: android.view.InflateException: Binary XML file line #126: Error inflating class <unknown>
at android.view.LayoutInflater.createView(LayoutInflater.java:645)
at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:764)
at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:704)
at android.view.LayoutInflater.rInflate(LayoutInflater.java:835)
at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:798)
at android.view.LayoutInflater.inflate(LayoutInflater.java:515)
... 23 more
Caused by: java.lang.reflect.InvocationTargetException
at java.lang.reflect.Constructor.newInstance(Native Method)
at android.view.LayoutInflater.createView(LayoutInflater.java:619)
... 28 more
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'android.graphics.drawable.Drawable android.graphics.drawable.Drawable$ConstantState.newDrawable()' on a null object reference
at android.graphics.drawable.LayerDrawable$ChildDrawable.<init>(LayerDrawable.java:1723)
at android.graphics.drawable.LayerDrawable$LayerState.<init>(LayerDrawable.java:1792)
at android.graphics.drawable.RippleDrawable$RippleState.<init>(RippleDrawable.java:998)
at android.graphics.drawable.RippleDrawable.createConstantState(RippleDrawable.java:989)
at android.graphics.drawable.RippleDrawable.createConstantState(RippleDrawable.java:988)
at android.graphics.drawable.LayerDrawable.mutate(LayerDrawable.java:1652)
at android.graphics.drawable.RippleDrawable.mutate(RippleDrawable.java:975)
at android.view.View.applyBackgroundTint(View.java:17432)
at android.view.View.setBackgroundDrawable(View.java:17308)
at com.google.android.material.floatingactionbutton.FloatingActionButton.access$101(FloatingActionButton.java:97)
at com.google.android.material.floatingactionbutton.FloatingActionButton$ShadowDelegateImpl.setBackgroundDrawable(FloatingActionButton.java:1384)
at com.google.android.material.floatingactionbutton.FloatingActionButtonImplLollipop.setBackgroundDrawable(FloatingActionButtonImplLollipop.java:80)
at com.google.android.material.floatingactionbutton.FloatingActionButton.<init>(FloatingActionButton.java:244)
at com.google.android.material.floatingactionbutton.FloatingActionButton.<init>(FloatingActionButton.java:193)
... 30 more
```

ovalenti, 2019-02-25:
Got the same with android 6 and version 2.0. However, I did a fresh build of 2.0.1 and got it working on the same device.

I guess that this got fixed as part of 3b74a8.

oliexdev, 2019-02-25:
see issue #405 or #406

closed 2019-02-25 by oliexdev