oliexdev/openScale #412: Mi composition scale 2 values are off and/or missing

prototux, 2019-02-25:
**Describe the bug**
The values from the xiaomi composition scale (scale v2) are off, sometime by a large value, some are missing:

* Muscle is 0%/0kg
* Bone mass is 0%/0kg
* Visceral fat is always 0
* Body water is off by more than 10%  (44.5% on mi fit vs 55.25% on openscale)
* Body fat % is a little off
* BMR is also off
* Protein percentage is missing (but it's an addition from last mi fit version)

It seems that mi fit uses a version of LibHTBodyFat, there's a project using it there: https://github.com/andychen83/iWellnessCN2 there's a jar lib HTBodyFat.jar but i guess it's only calling the JNIs, which are available for multiple architectures.

(version 2.0, from f-droid)

oliexdev, 2019-02-25:
that's correct which is also written in the [openScale wiki](https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale)!
The mi scale v2 is using a `libBodyfat.so` for the calculation. You are welcome to reverse engineer this library.

prototux, 2019-02-25:
I've understood that this library was reverse engineered and implemented already in`/android_app/app/src/main/java/com/health/openscale/core/bluetooth/lib/OneByoneLib.java` (from issue #159 the one you redirect issues about mi composition scale to, like in #268 ), unless holtek makes multiple version of their libBodyFat.so one for each mcu, there's no real reason to not use the OneByoneLib.java for the mi composition scale i think, as the impedance to data calculations should be the same.

oliexdev, 2019-02-27:
Unfortunately, they are **not** the same.

prototux, 2019-03-01:
I'm currently reverse engineering the last version of the libBodyfat.so that comes with mi fit (v 3.5.8), that new version comes with some new features compared to the previous versions:
* This new one also calculate the ideal weight (but it's kinda "broken" as it's only doing a reverse BMI using BMI 22.0 (eg. (22.0f*heightCm)*heightCm/10000) instead of a proper formula
* This one also create the "rating tables", but it's not used by mi fit as they did it themselves before already (and the ones from holtek isn't as precise, for example it only have "slim/normal/overweight/obese" for BMI where mi fit uses obese 1 and obese 2
* Beside that, the .so is still not really complicated, it's only a bunch of algorithms that they put in a JNI to seemingly "avoid" their formulas being decompiled from the jar that comes with the JNI.
* It does **not** however calculate the protein percentage, that is done by mi fit itself it seems.
* It does not also calculate the "body age", even if the jar lib does have a (unused) attribute for that.

Going to the mi scale 2 itself, it doesn't seem that openscale does much with it, and more surpinsingly, it seems to connect and wait for notifications where the scale actually uses ad packets to send the weight, but, i don't find where the impedance could be, as the current ad packet only have 16 bits of free space (and they seem, along with a bit flag, to change when the scale get the impedance!) where it seems impedance needs at least 24 bits.
If the impedance really is on 16bits and the same format as the weight, i would have an impedance of 429, it seems low.

You've repeatedly said that the impedance is sent along the weight with the ad packet, is it these 16 bits or am i missing something here? (the adtype 225 packet doesn't change at all)

prototux, 2019-03-02:
I've successfully reverse engineered the libBodyfat.so that is in mi fit v3.5.8, you can find a (currently dirty but working, will work on refactoring later) python implementation [here](https://github.com/prototux/MIBCS-reverse-engineering/blob/master/calculations.py)
The 2 bytes actually are the impedance, it's the same format as the weight.

oliexdev, 2019-03-04:
thanks to @prototux we have the calculations for the mi scale v2 body measurements with commit https://github.com/oliexdev/openScale/commit/d2678fca853343b86141e976150c65c940f05797

closed 2019-03-04 by oliexdev

Extarys, 2019-10-16:
Hi folks! I just got my Mi Composition Scale v2 an hour ago and I downloaded the dev version of the app. Thanks for reverse engineering the scale. Please tell me if I should open a new issue for this.

I'm 5'8" and weight ~155 lbs. I'm quite sedentary as I'm sick but I'm not overweight (maybe a couple of lbs)

[image]

On this screenshot:
- 80% of muscle, which is way to high, also the original app shows this in lbs or kgs
- 46.51 % of body fat, from my understanding, over 27 % is quite high (overweight and more?)
- Water may be ok as I just woke up and was very thirsty
- Visceral fat is missing compared to original app

How can I help improve the accuracy of the app? Also note that this is the second or third time I played with it in the hour and it may read better results eventually? I'm no engineering but I can help with logs or data or whatever.

I did not install the official Mi app or did any firmware update (if there is any)