oliexdev/openScale #414: Turning off lines from measurement bar in overview page

ivoid, 2019-02-25:
If all the lines in the graph are turned off using the measurement bar then there's no way to bring them back. X-axis is shown but Y-axis is blank and no graph appears.

Thanks and keep up the good work.

oliexdev, 2019-02-25:
I don't really understand you can bring it back if you enable it again!?

ivoid, 2019-02-25:
It only happens if rolling chart option is enabled.
My normal overview page.

[image]

All lines turned off.

[image]

Turn back on, no lines appear. The lines will only re-appear after toggling any of the options in the graph's display option menu.

[image]

oliexdev, 2019-02-27:
can't reproduce it. Did you test it with the latest dev version?

closed 2019-03-17 by oliexdev