oliexdev/openScale #415: Data import seems to be broken.

WinMinerGMX, 2019-02-25:
I wasn't happy with how v2.0 displays the graphs so I wanted to downgrade to an older version which still has the graph scaling that I like. So I exported the data (openScale User.csv) through the export menu and downgraded to v1.9.3 through Fdroid. Import of data doesn't seem to do anything. Chart says: No chart data available and the user doesn't get recreated. So I tried to import with v2.0 again. Same issue. No Info in FAQ, Wiki, Issue-Tracker or anywhere else.

Any help?

WinMinerGMX, 2019-02-25:
Importing with user already created yields the error message displayed in the screenshot.

[image]

oliexdev, 2019-02-25:
good observation :wink:

closed 2019-02-25 by oliexdev