oliexdev/openScale #416: BT connection to Beurer BF710 broken in Version 2.0 and 2.0.1

reklawykSekuL, 2019-02-25:
Errors occurred when connecting to the BF710
This occurred after updating the openScale app for Android from Version 1.8.4  to  2.0 and also 2.0.1

I've Tried to add  the log file (36,2KiB) - but it failed with "Something went really wrong. and we can't process that file. Try again

I've Tried to add  the log file as Tar-Archive (4,1KiB) - but it failed with "Something went really wrong. and we can't process that file. Try again

start of the log: 
`2019-02-25 19:32:01.423 Debug [2] AboutPreferences: Debug log enabled, openScale v2.0.1 (42), SDK 26, motorola Moto Z2 Play
2019-02-25 19:32:01.430 Debug [2] AboutPreferences: Selected user id(1) name(Nxxxxxx) birthday(Thu May 03 00:00:00 GMT+01:00 19xx) age(xx) body height(xxx,00) scale unit(kg) gender(male) initial weight(xxx,00) goal weight(90,00) goal date(Wed Jul 03 18:34:08 GMT+02:00 2019) measure unt(cm) activity level(0)
2019-02-25 19:32:08.423 Debug [2] OpenScale: Trying to connect to bluetooth device [84:EB:18:5B:63:96] (Beurer BF710)
2019-02-25 19:32:08.425 Debug [2] BluetoothCommunication: No location permission, connecting without LE scan
2019-02-25 19:32:08.927 Debug [2] BluetoothCommunication: Try to connect to BLE device 84:EB:18:5B:63:96
2019-02-25 19:32:08.929 Debug [2] BluetoothCommunication: Step Nr 0
2019-02-25 19:32:08.931 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe1
2019-02-25 19:32:08.946 Debug [32408] RxBle#ClientOperationQueue: QUEUED   ConnectOperation(227272824)
2019-02-25 19:32:08.948 Debug [32400] RxBle#ClientOperationQueue: STARTED  ConnectOperation(227272824)
2019-02-25 19:32:21.110 Debug [32406] RxBle#BluetoothGatt$1: onConnectionStateChange newState=2 status=0
2019-02-25 19:32:21.119 Debug [32400] RxBle#ClientOperationQueue: FINISHED ConnectOperation(227272824) in 12170 ms
2019-02-25 19:32:21.128 Debug [2] MainActivity: Bluetooth connection successful established
2019-02-25 19:32:21.155 Debug [2] RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(173343042)
2019-02-25 19:32:21.194 Debug [32411] RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(173343042)
2019-02-25 19:32:22.296 Debug [32406] RxBle#BluetoothGatt$1: onServicesDiscovered status=0
2019-02-25 19:32:22.302 Debug [32408] RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(54480528)
2019-02-25 19:32:22.303 Debug [32411] RxBle#ConnectionOperationQueue: FINISHED ServiceDiscoveryOperation(173343042) in 1109 ms
2019-02-25 19:32:22.304 Debug [32411] RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(54480528)
2019-02-25 19:32:22.332 Debug [32406] RxBle#BluetoothGatt$1: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
2019-02-25 19:32:22.334 Debug [32411] RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(54480528) in 31 ms
2019-02-25 19:32:22.335 Debug [32408] BluetoothCommunication: Successful set notification on for 0xffe1
2019-02-25 19:32:22.336 Debug [32408] BluetoothCommunication: Step Nr 1
2019-02-25 19:32:22.337 Debug [32408] BluetoothCommunication: Invoke write bytes [E6 01] on 0xffe1
2019-02-25 19:32:22.340 Debug [32408] RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(214463407)
2019-02-25 19:32:22.341 Debug [32411] RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(214463407)
2019-02-25 19:32:22.342 Debug [32408] BluetoothCommunication: Stop machine state
2019-02-25 19:32:27.390 Debug [32406] RxBle#BluetoothGatt$1: onCharacteristicWrite characteristic=0000ffe1-0000-1000-8000-00805f9b34fb status=133
2019-02-25 19:32:27.395 Debug [32411] RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(214463407) in 5053 ms
2019-02-25 19:32:27.399 Debug [2] RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(417704)
2019-02-25 19:32:27.400 Debug [32406] RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=8
2019-02-25 19:32:27.409 Debug [32411] RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(417704)
2019-02-25 19:32:27.412 Info [32406] RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (84:EB:18:5B:63:96)
2019-02-25 19:32:27.417 Warning [32411] RxBle#ConnectionOperationQueue: Queue's awaitRelease() has been interrupted abruptly while it wasn't released by the release() method.
java.lang.InterruptedException
	at java.lang.Object.wait(Native Method)
	at com.polidea.rxandroidble2.internal.serialization.QueueSemaphore.awaitRelease(QueueSemaphore.java:15)
	at com.polidea.rxandroidble2.internal.serialization.ConnectionOperationQueueImpl$1.run(ConnectionOperationQueueImpl.java:71)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:457)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
	at java.lang.Thread.run(Thread.java:764)

2019-02-25 19:32:27.437 Debug [32406] RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(248326567)
2019-02-25 19:32:27.438 Error [2] MainActivity: No Bluetooth device found retrying
2019-02-25 19:32:27.438 Debug [32400] RxBle#ClientOperationQueue: STARTED  DisconnectOperation(248326567)
2019-02-25 19:32:27.443 Debug [32411] RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(417704) in 42 ms
2019-02-25 19:32:27.444 Debug [32411] RxBle#Executors$RunnableAdapter: Terminated.
2019-02-25 19:32:27.450 Debug [32400] RxBle#ClientOperationQueue: FINISHED DisconnectOperation(248326567) in 12 ms
2019-02-25 19:32:27.460 Debug [32408] RxBle#ClientOperationQueue: QUEUED   ConnectOperation(163797054)
2019-02-25 19:32:27.461 Debug [32400] RxBle#ClientOperationQueue: STARTED  ConnectOperation(163797054)
2019-02-25 19:32:53.237 Debug [32395] RxBle#BluetoothGatt$1: onConnectionStateChange newState=2 status=0
2019-02-25 19:32:53.244 Debug [32400] RxBle#ClientOperationQueue: FINISHED ConnectOperation(163797054) in 25783 ms
2019-02-25 19:32:53.248 Debug [2] MainActivity: Bluetooth connection successful established
2019-02-25 19:32:53.259 Debug [2] RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(154473397)
2019-02-25 19:32:53.260 Debug [32411] RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(154473397)
2019-02-25 19:32:54.412 Debug [32395] RxBle#BluetoothGatt$1: onServicesDiscovered status=0
2019-02-25 19:32:54.423 Debug [32408] RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(145764630)
2019-02-25 19:32:54.426 Debug [32408] RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(251513495)
2019-02-25 19:32:54.429 Debug [32411] RxBle#ConnectionOperationQueue: FINISHED ServiceDiscoveryOperation(154473397) in 1169 ms
2019-02-25 19:32:54.430 Debug [32411] RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(145764630)
2019-02-25 19:32:54.482 Debug [32406] RxBle#BluetoothGatt$1: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
[... 24 lines cut ...]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
	at java.lang.Thread.run(Thread.java:764)

2019-02-25 19:32:59.518 Debug [32395] RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(66865658)
2019-02-25 19:32:59.519 Debug [32400] RxBle#ClientOperationQueue: STARTED  DisconnectOperation(66865658)
2019-02-25 19:32:59.520 Error [2] MainActivity: No Bluetooth device found retrying
2019-02-25 19:32:59.522 Debug [2] RxBle#ConnectionOperationQueue$2: REMOVED  CharacteristicWriteOperation(118838512)
2019-02-25 19:32:59.523 Debug [2] RxBle#ConnectionOperationQueue$2: REMOVED  CharacteristicWriteOperation(26398748)
2019-02-25 19:32:59.525 Debug [32411] RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(123798637) in 21 ms
2019-02-25 19:32:59.529 Debug [32400] RxBle#ClientOperationQueue: FINISHED DisconnectOperation(66865658) in 9 ms
2019-02-25 19:32:59.530 Debug [32411] RxBle#Executors$RunnableAdapter: Terminated.
2019-02-25 19:32:59.545 Debug [32408] RxBle#ClientOperationQueue: QUEUED   ConnectOperation(223929053)
2019-02-25 19:32:59.548 Debug [32400] RxBle#ClientOperationQueue: STARTED  ConnectOperation(223929053)
2019-02-25 19:32:59.559 Error [2] MainActivity: Bluetooth unexpected error: Disconnected from 84:EB:18:5B:63:96 with status 8 (GATT_INSUF_AUTHORIZATION)
[... cut ...]
2019-02-25 19:33:08.962 Debug [32400] RxBle#ClientOperationQueue: STARTED  DisconnectOperation(128175518)
2019-02-25 19:33:08.963 Debug [32411] RxBle#Executors$RunnableAdapter: Terminated.
2019-02-25 19:33:08.968 Debug [2] MainActivity: Bluetooth connection successful disconnected
2019-02-25 19:33:08.971 Debug [32400] RxBle#ClientOperationQueue: FINISHED DisconnectOperation(128175518) in 8 ms
2019-02-25 19:33:41.262 Debug [2] OpenScale: Disconnecting from bluetooth device
2019-02-25 19:33:41.264 Debug [2] BluetoothCommunication: Bluetooth disconnect`

oliexdev, 2019-03-10:
could you post the openScale log file if you have disable your GPS and please use proper the markdown formatting if you post some logs.

reklawykSekuL, 2019-03-13:
i've made a new log with Google Location Settings (Standortmodus)  Mode set to power-safe (Stromsparfunktion) - only in this Setting i found GPS mentioned.

the first connection try fails again - but the second try worked.
that works much better than before, but have you noticed that  the [bluetooth library](https://github.com/Polidea/RxAndroidBle/releases) has a higher version now : 1.8.1-rxjava2

here comes the log:

```2019-03-13 18:40:22.854 Debug [2] AboutPreferences: Debug log enabled, openScale v2.0.1 (42), SDK 26, motorola Moto Z2 Play
2019-03-13 18:40:22.855 Debug [2] AboutPreferences: Selected user id(1) name(Nor****) birthday(*** *** ** 00:00:00 GMT+01:00 19**) age(**) body height(***,00) scale unit(kg) gender(male) initial weight(***,00) goal weight(**,**) goal date(Wed Jul 03 18:34:08 GMT+02:00 2019) measure unt(cm) activity level(0)
2019-03-13 18:40:29.300 Debug [2] OpenScale: Trying to connect to bluetooth device [84:EB:18:5B:63:96] (Beurer BF710)
2019-03-13 18:40:29.314 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2019-03-13 18:40:29.345 Debug [2] RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(158123290)
2019-03-13 18:40:29.346 Debug [3424] RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(158123290)
2019-03-13 18:40:29.348 Info [3431] RxBle#QueueOperation: Scan operation is requested to start.
2019-03-13 18:40:29.361 Debug [3424] RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(158123290) in 16 ms
2019-03-13 18:40:48.627 Debug [2] BluetoothCommunication: Stop Le san
2019-03-13 18:40:48.632 Info [3431] RxBle#CancellableDisposable: Scan operation is requested to stop.
2019-03-13 18:40:49.136 Debug [2] BluetoothCommunication: Try to connect to BLE device 84:EB:18:5B:63:96
2019-03-13 18:40:49.145 Debug [2] BluetoothCommunication: Step Nr 0
2019-03-13 18:40:49.147 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe1
2019-03-13 18:40:49.178 Debug [3433] RxBle#ClientOperationQueue: QUEUED   ConnectOperation(194029770)
2019-03-13 18:40:49.179 Debug [3424] RxBle#ClientOperationQueue: STARTED  ConnectOperation(194029770)
2019-03-13 18:40:53.887 Debug [3419] RxBle#BluetoothGatt$1: onConnectionStateChange newState=2 status=0
2019-03-13 18:40:53.891 Debug [3424] RxBle#ClientOperationQueue: FINISHED ConnectOperation(194029770) in 4712 ms
2019-03-13 18:40:53.900 Debug [2] MainActivity: Bluetooth connection successful established
2019-03-13 18:40:53.922 Debug [2] RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(182572721)
2019-03-13 18:40:53.924 Debug [3432] RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(182572721)
2019-03-13 18:40:55.092 Debug [3419] RxBle#BluetoothGatt$1: onServicesDiscovered status=0
2019-03-13 18:40:55.098 Debug [3433] RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(252456173)
2019-03-13 18:40:55.099 Debug [3432] RxBle#ConnectionOperationQueue: FINISHED ServiceDiscoveryOperation(182572721) in 1176 ms
2019-03-13 18:40:55.100 Debug [3432] RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(252456173)
2019-03-13 18:40:55.129 Debug [3419] RxBle#BluetoothGatt$1: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
2019-03-13 18:40:55.132 Debug [3432] RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(252456173) in 31 ms
2019-03-13 18:40:55.134 Debug [3433] BluetoothCommunication: Successful set notification on for 0xffe1
2019-03-13 18:40:55.137 Debug [3433] BluetoothCommunication: Step Nr 1
2019-03-13 18:40:55.140 Debug [3433] BluetoothCommunication: Invoke write bytes [E6 01] on 0xffe1
2019-03-13 18:40:55.146 Debug [3433] RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(190248816)
2019-03-13 18:40:55.147 Debug [3432] RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(190248816)
2019-03-13 18:40:55.147 Debug [3433] BluetoothCommunication: Stop machine state
2019-03-13 18:41:00.164 Debug [3419] RxBle#BluetoothGatt$1: onCharacteristicWrite characteristic=0000ffe1-0000-1000-8000-00805f9b34fb status=133
2019-03-13 18:41:00.178 Debug [3428] RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=8
2019-03-13 18:41:00.179 Debug [3432] RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(190248816) in 5032 ms
2019-03-13 18:41:00.181 Debug [2] RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(74124453)
2019-03-13 18:41:00.182 Info [3428] RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (84:EB:18:5B:63:96)
2019-03-13 18:41:00.183 Debug [3432] RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(74124453)
2019-03-13 18:41:00.185 Debug [3428] RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(202168442)
2019-03-13 18:41:00.190 Error [2] MainActivity: No Bluetooth device found retrying
2019-03-13 18:41:00.190 Warning [3432] RxBle#ConnectionOperationQueue: Queue's awaitRelease() has been interrupted abruptly while it wasn't released by the release() method.
java.lang.InterruptedException
	at java.lang.Object.wait(Native Method)
	at com.polidea.rxandroidble2.internal.serialization.QueueSemaphore.awaitRelease(QueueSemaphore.java:15)
	at com.polidea.rxandroidble2.internal.serialization.ConnectionOperationQueueImpl$1.run(ConnectionOperationQueueImpl.java:71)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:457)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[... cut ...]
2019-03-13 18:42:42.135 Debug [3433] RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(158716318)
2019-03-13 18:42:42.137 Debug [3424] RxBle#ClientOperationQueue: STARTED  DisconnectOperation(158716318)
2019-03-13 18:42:42.151 Debug [3418] RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=0
2019-03-13 18:42:42.153 Debug [2] MainActivity: Bluetooth connection successful disconnected
2019-03-13 18:42:42.156 Debug [3424] RxBle#ClientOperationQueue: FINISHED DisconnectOperation(158716318) in 19 ms```
-

jroth-haj, 2019-03-15:
Also a Bluetooth Problem with SANITAS SBF70 since update from 2.0 to 2.01 (42)

oliexdev, 2019-03-17:
I will close this issue because it is highly related to issue #433
Please comment on this issue for further related problems.

closed 2019-03-17 by oliexdev