oliexdev/openScale #417: Sensun scale

quthla, 2019-02-26:
When weighing one time, since latest version it'll insert dozens of measurements with all the same values.

gmarco, 2019-02-26:
i'll take a look at it next days.there was changed a lot.

voyager, 2019-03-03:
Just to illustrate what @quthla referred to:

[image]

quthla, 2019-03-04:
> 
> 
> Just to illustrate what @quthla referred to:
> 
> [image]

That's exactly it. Also the values are not all the same. The first few measurements have partially old values but following measurements then only have current values.

voyager, 2019-03-04:
It seems that the app does not consolidate the feed and then enters the last value, but instead saves all the incoming values, almost like it's saving raw feed, as now one can see the table being live-fed and expanded with new rows while the app is connected to the scale.

quthla, 2019-03-04:
@oliexdev Is this caused by your new BT state machine?

gmarco, 2019-03-04:
yes the last changes break the function. i'll try to fix this today.
The Scale is sending continous the weight (thats why the many saves)

IronJunkie, 2019-03-10:
Sorry, above ref is not valid to this issue.

closed 2019-03-17 by oliexdev