oliexdev/openScale #419: Please reconsider graph display in Version v2.0.1

Galopperfan, 2019-02-26:
Hi,
I was very happy with this app (working really fantastic with my Sanitas SBF70), very fast connection after the penultimate update.
BUT, the graph display in the new Version 2.0.1. is simply unreadable without a magnifying glass.
Perhaps, I've not got it how to enlarge the view, but it is impossible for me to read the weight of the last couple of days in the graph display. The "points" to "click in" are really tiny. The display of my Aquaris X5 plus is 5 inches, so it is no "small" phone.
Please reconsider the graph display or implement an option in the settings to get the "old view" back.
That would be great!
Best regards,
Max

rainMakerBC, 2019-02-27:
Hi Max,

I agree.

I can't read it as well. I suggest reverting back to the number display for the averages.

Thanks.

RM

oliexdev, 2019-03-10:
Well, I don't go back because the old graph library are not supported anymore and with the new graph we have some advantages for example a marker view, rolling chart and so on. Additionally, I did a lot of effort to update the library which only thank with bad reviews :disappointed: 
I had this new graph long time as a branch version and everybody could test it and contribute.
I also don't understand what is now unreadable? The text size is the same as the old graph!? I only added some additionally information, which is a bit smaller (the date and the difference indicator). 
Also you can even click better the points because you don't have to be exactly on the point anymore!?
I could go on, many improvements was made.
But openScale is open-source so you could propose with a PR your changes. Then I could review your PR and maybe I will merge it into openScale.

closed 2019-03-10 by oliexdev

gappleto97, 2019-03-26:
The problem is the difference in scale. The new graph shows the goal line and zooms out accordingly. It actually zooms out even more than the dedicated graph view. The old graph would zoom such that the top and bottom weights were at the edges, so you could see the movement better.