oliexdev/openScale #420: Broken notification

SriramKeerthi, 2019-02-27:
My SBF75 doesn't sync with the app anymore. Bluetooth connection gets established correctly, but when it's about to sync the measurement, I get an error notification.

Seems like a resource is missing during localisation? But I want the scale in English, don't remember ever changing that. Here's the screenshot: https://imgur.com/a/ElnnDX4

SriramKeerthi, 2019-02-27:
I tried deleting the app and installing it again, didn't help

runyalfa, 2019-03-07:
Hi,
I have the same issue since the last updated

oliexdev, 2019-03-10:
please post probably the openScale log file

runyalfa, 2019-03-11:
19-03-10 19:36:23.982 Debug [2] AboutPreferences: Debug log enabled, openScale v2.0.1 (42), SDK 26, Xiaomi MI 6
2019-03-10 19:36:23.983 Debug [2] AboutPreferences: Selected user id(1) name(Runy) birthday(Fri Dec 06 00:00:00 GMT+01:00 1974) age(44) body height(178,00) scale unit(kg) gender(male) initial weight(83,10) goal weight(75,00) goal date(Tue May 07 15:54:42 GMT+02:00 2019) measure unt(cm) activity level(2)
2019-03-10 19:36:43.750 Debug [2] BluetoothCommunication: Stop Le san
2019-03-10 19:36:43.757 Info [10208] RxBle#CancellableDisposable: Scan operation is requested to stop.
2019-03-10 19:36:44.268 Debug [2] BluetoothCommunication: Try to connect to BLE device F4:B8:5E:DF:99:B3
2019-03-10 19:36:44.270 Debug [2] BluetoothCommunication: Step Nr 0
2019-03-10 19:36:44.271 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe1
2019-03-10 19:36:44.287 Debug [10214] RxBle#ClientOperationQueue: QUEUED   ConnectOperation(37467794)
2019-03-10 19:36:44.289 Debug [10204] RxBle#ClientOperationQueue: STARTED  ConnectOperation(37467794)
2019-03-10 19:36:54.082 Debug [10210] RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=133
2019-03-10 19:36:54.095 Info [10210] RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (F4:B8:5E:DF:99:B3)
2019-03-10 19:36:54.096 Debug [10213] RxBle#Executors$RunnableAdapter: Terminated.
2019-03-10 19:36:54.098 Debug [10204] RxBle#ClientOperationQueue: FINISHED ConnectOperation(37467794) in 9809 ms
2019-03-10 19:36:54.102 Debug [10210] RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(110218336)
2019-03-10 19:36:54.103 Debug [10204] RxBle#ClientOperationQueue: STARTED  DisconnectOperation(110218336)
2019-03-10 19:36:54.112 Debug [10204] RxBle#ClientOperationQueue: FINISHED DisconnectOperation(110218336) in 8 ms
2019-03-10 19:36:54.119 Error [2] MainActivity: No Bluetooth device found retrying
2019-03-10 19:36:54.131 Debug [10214] RxBle#ClientOperationQueue: QUEUED   ConnectOperation(78217435)
2019-03-10 19:36:54.132 Debug [10204] RxBle#ClientOperationQueue: STARTED  ConnectOperation(78217435)
2019-03-10 19:37:03.958 Debug [10210] RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=133
2019-03-10 19:37:03.960 Info [10210] RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (F4:B8:5E:DF:99:B3)
2019-03-10 19:37:03.961 Debug [10213] RxBle#Executors$RunnableAdapter: Terminated.
2019-03-10 19:37:03.962 Debug [10210] RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(141301430)
2019-03-10 19:37:03.965 Debug [10204] RxBle#ClientOperationQueue: FINISHED ConnectOperation(78217435) in 9832 ms
2019-03-10 19:37:03.966 Debug [10204] RxBle#ClientOperationQueue: STARTED  DisconnectOperation(141301430)
2019-03-10 19:37:03.971 Debug [10204] RxBle#ClientOperationQueue: FINISHED DisconnectOperation(141301430) in 5 ms
2019-03-10 19:37:03.973 Error [2] MainActivity: No Bluetooth device found retrying
2019-03-10 19:37:03.982 Debug [10214] RxBle#ClientOperationQueue: QUEUED   ConnectOperation(164943497)
2019-03-10 19:37:03.983 Debug [10204] RxBle#ClientOperationQueue: STARTED  ConnectOperation(164943497)
2019-03-10 19:37:08.977 Debug [10210] RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=133
2019-03-10 19:37:08.980 Info [10210] RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (F4:B8:5E:DF:99:B3)
2019-03-10 19:37:08.982 Debug [10204] RxBle#ClientOperationQueue: FINISHED ConnectOperation(164943497) in 4998 ms
2019-03-10 19:37:08.983 Debug [10213] RxBle#Executors$RunnableAdapter: Terminated.
2019-03-10 19:37:08.984 Debug [10210] RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(150041276)
2019-03-10 19:37:08.986 Debug [10204] RxBle#ClientOperationQueue: STARTED  DisconnectOperation(150041276)
2019-03-10 19:37:08.991 Debug [10204] RxBle#ClientOperationQueue: FINISHED DisconnectOperation(150041276) in 6 ms
2019-03-10 19:37:08.992 Error [2] MainActivity: No Bluetooth device found retrying
2019-03-10 19:37:09.001 Debug [10214] RxBle#ClientOperationQueue: QUEUED   ConnectOperation(142840999)
2019-03-10 19:37:09.002 Debug [10204] RxBle#ClientOperationQueue: STARTED  ConnectOperation(142840999)
2019-03-10 19:37:13.982 Debug [10210] RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=133
2019-03-10 19:37:13.984 Info [10210] RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (F4:B8:5E:DF:99:B3)
2019-03-10 19:37:13.985 Debug [10213] RxBle#Executors$RunnableAdapter: Terminated.
2019-03-10 19:37:13.985 Debug [10210] RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(149849341)
2019-03-10 19:37:13.986 Debug [10204] RxBle#ClientOperationQueue: FINISHED ConnectOperation(142840999) in 4983 ms
2019-03-10 19:37:13.989 Debug [10204] RxBle#ClientOperationQueue: STARTED  DisconnectOperation(149849341)
2019-03-10 19:37:13.992 Error [2] MainActivity: No Bluetooth device found retrying
2019-03-10 19:37:13.994 Debug [10204] RxBle#ClientOperationQueue: FINISHED DisconnectOperation(149849341) in 8 ms
2019-03-10 19:37:13.999 Error [2] MainActivity: Bluetooth unexpected error: Disconnected from F4:B8:5E:DF:99:B3 with status 133 (GATT_ERROR)
2019-03-10 19:42:32.149 Debug [2] BluetoothCommunication: Timeout Bluetooth disconnect
2019-03-10 19:42:32.151 Debug [2] BluetoothCommunication: Bluetooth disconnect
2019-03-10 19:42:32.164 Debug [2] MainActivity: Bluetooth connection successful disconnected

oliexdev, 2019-03-17:
I will close this issue because it is highly related to issue #433
Please comment on this issue for further related problems.

closed 2019-03-17 by oliexdev