oliexdev/openScale #422: missing statistics in statistics-tab

john-bill, 2019-03-02:
Hi,
I really would like to get the weight-average statistics back in v2.x.
Comparing weight-per-week to weight-per-month was on of the best features in v1.x, now I have two (at least for me) unreadable diagrams without the statistics information.
Is there a chance to get this (and maybe some other statistics) back?

oliexdev, 2019-03-10:
click on the diagram then you will see the values!?

closed 2019-03-10 by oliexdev

john-bill, 2019-03-10:
in both diagrams (which look exactly the same) I only get the exact same actual data, nothing per week or per month.

oliexdev, 2019-03-10:
then your average for last month and week are the same

oliexdev, 2019-03-10:
both diagram are compared with your last measurement

john-bill, 2019-03-10:
average for month and for week are different! But this is not reflected in the diagrams nor do I see anything besides the actual measurement in it.

gappleto97, 2019-03-26:
Agreed, this seems to be a bug. My weekly should be 182.3, yet it shows 181.6, which happens to be the most recent measurement, and also what monthly shows.