oliexdev/openScale #424: Crash on Bluetooth

jimbolaya, 2019-03-03:
**Describe the bug**
When bluetooth is on, openscale crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Start openscale from screen
2. It offers to start bluetooth. I say accept
3. It crashes until bluetooth is turned off.
4. It does the same thing if bluetooth is already on.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): 
No

**Expected behavior**
I was hoping it wouldn't crash

**Debug log**
Build version: 1.9.3 
Build date: 1979-11-30 00:00:00 
Current date: 2019-03-03 10:49:13 
Device: Motorola XT1080 
 
Stack trace:  
java.lang.SecurityException: "gps" location provider requires ACCESS_FINE_LOCATION permission.
	at android.os.Parcel.readException(Parcel.java:1472)
	at android.os.Parcel.readException(Parcel.java:1426)
	at android.location.ILocationManager$Stub$Proxy.isProviderEnabled(ILocationManager.java:948)
	at android.location.LocationManager.isProviderEnabled(LocationManager.java:1117)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.connect(BluetoothCommunication.java:487)
	at com.health.openscale.core.OpenScale.connectToBluetoothDevice(OpenScale.java:599)
	at com.health.openscale.gui.MainActivity.invokeConnectToBluetoothDevice(MainActivity.java:462)
	at com.health.openscale.gui.MainActivity.onCreateOptionsMenu(MainActivity.java:409)
	at android.app.Activity.onCreatePanelMenu(Activity.java:2546)
	at androidx.fragment.app.FragmentActivity.onCreatePanelMenu(FragmentActivity.java:378)
	at androidx.appcompat.view.WindowCallbackWrapper.onCreatePanelMenu(WindowCallbackWrapper.java:94)
	at androidx.appcompat.app.AppCompatDelegateImpl$AppCompatWindowCallback.onCreatePanelMenu(AppCompatDelegateImpl.java:2549)
	at androidx.appcompat.view.WindowCallbackWrapper.onCreatePanelMenu(WindowCallbackWrapper.java:94)
	at androidx.appcompat.app.ToolbarActionBar.populateOptionsMenu(ToolbarActionBar.java:455)
	at androidx.appcompat.app.ToolbarActionBar$1.run(ToolbarActionBar.java:56)
	at android.os.Handler.handleCallback(Handler.java:733)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:136)
	at android.app.ActivityThread.main(ActivityThread.java:5097)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:515)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:785)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:601)
	at dalvik.system.NativeStart.main(Native Method)

oliexdev, 2019-03-10:
> Reproduced with latest dev version:
> no

would resolve your problem

closed 2019-03-10 by oliexdev

jimbolaya, 2019-03-10:
Is there a way for me to see the data from my AppStore version in the dev version? 

I seem to recall they keep the data in different places.

Will I always need to use the dev version going forward?

oliexdev, 2019-03-10:
the latest version on GooglePlay (version 2.0.1) would also fixed your problem.