oliexdev/openScale #425: openScale syncs only the last measurement

steffenl82, 2019-03-04:
Hi 

I'm using a Medisana BS444 scale together with openScale which works just fine.

I'm *not* syncing the scale with openScale every time I step on the scale.

Prior to version 2 (i.e. <=1.9.3) openScale was pulling *all* the data since the last sync. However, since version 2 only the current/last measurement seems to be synced into the app.

Is that an intended behavior?

oliexdev, 2019-03-10:
this is not possible because your scale doesn't support any history data.

closed 2019-03-10 by oliexdev

steffenl82, 2019-03-10:
Your statement is certainly not correct.

As written in my initial post, I am able to pull multiple/historical measurements off the scale with openScale 1.9.3 - but this is not working since V2

lbschenkel, 2019-04-08:
Hi, I am considering purchasing a BS444. After I checked the manual online and some Googling I was not able to find out how many measurements it stores on its internal memory. I found this ticket and now I'm confused about the BS444 capabilities, given the contradictory reports. I would highly appreciate if somebody owning the scale could clarify the following, so I can make an informed purchase decision:

1. how many measurements are stored in the scale? just the last one, or more?
2. is the answer to question 1 per user, or global? in case of 2 users, would user 2 overwrite the measurements of user 1?
3. do I need to have openScale open when stepping on the scale or can I sync after the fact, away from the scale?

Thank you.

steffenl82, 2019-04-08:
Hi 

I may answer some of your questions at least in part.

First of all: I own a Medisana BS444.

Regarding 1: The scale definitely stores more than one measurement. I don't know the maximum, but one time I was able to pull off 6 or so old measurements. However, as indicated in my initial post, this only works with openscale prior to version 2 (I downgraded to 1.9.3 exactly for that reason/functionality). In version 2 (I tested 2.0.0 and 2.0.1 - I never tried 2.0.2 or 2.0.3) openscale pulls the last measurement only.

Regarding 2: I cannot help you, because I never tried a second user. I don't know (or even see a way) how to set up several users ''directly'' at the scale. My impression is that the scale just collects the data (not knowing of any users at all) and the actual distinction between the users is done by openscale. 

Regarding 3: When you step on the scale, the scale automatically turns on and collects a measurement. When the measurement has been done, the corresponding data is shown in the display of the scale (i.e. the display automatically and repeatedly flips through several pages with different values: water, fat, etc). You may step off the scale at this time. After about 10 or 15 seconds the scale automatically turns off. As far as I can see, the scale is in 'transmission mode' only during that short time when it displays the measured values (this is also the only time when I can see a bluetooth symbol in the display). This means that syncing with openscale can only be done during that time period.

lbschenkel, 2019-04-08:
Thank you very much. Comments inline.

> Regarding 1: The scale definitely stores more than one measurement. I don't know the maximum, but one time I was able to pull off 6 or so old measurements. However, as indicated in my initial post, [...]

That's a bummer. It's puzzling that you can retrieve the history using an old version of openScale and yet @oliexdev who wrote the software says the scale doesn't store history. How's that possible?

> Regarding 2: I cannot help you, because I never tried a second user. I don't know (or even see a way) how to set up several users ''directly'' at the scale. My impression is that the scale just [...]

According to the PDF manual (https://www.medisana.com/out/pictures/media/manual/bs430_bs440qsg.pdf) which can be downloaded from the product page (https://www.medisana.com/en/Health-control/Personal-scales/BS-444-connect-Body-analysis-scale.html), the scale supports up to 8 users and it contains the instructions on how to set them up in the scale (English instructions on page 4).

> Regarding 3: When you step on the scale, the scale automatically turns on and collects a measurement. When the measurement has been done, the corresponding data is shown in the display of the scale [...]

I see. I was hoping that I wouldn't need to take my phone with me when weighting myself, but it looks like that won't be possible with this scale.

Thanks again for taking your time to answer my questions.

steffenl82, 2019-04-08:
Thank you for the link with the manual!

> That's a bummer. It's puzzling that you can retrieve the history using an old version of openScale and yet @oliexdev who wrote the software says the scale doesn't store history. How's that possible?

I think he just does not own each and every scale that his app supports. So he gave an uninformed answer. The manual now even gives the proof that historic values are supported by that scale, because it explicitly states that "The scales can store the data (gender, athlete mode, height, age) for 8 people with up to 30 memory locations each." -> this is also the answer to your initial question: 30 measurements per user.

> I see. I was hoping that I wouldn't need to take my phone with me when weighting myself, but it looks like that won't be possible with this scale.

At least you don't have to take your phone with you every time you are doing a measurement. Because if you don't sync during that 15 seconds period, the measurement is saved and can be retrieved the next time.

lbschenkel, 2019-04-08:
> "The scales can store the data (gender, athlete mode, height, age) for 8 people with up to 30 memory locations each." -> this is also the answer to your initial question: 30 measurements per user.

Where did you find this? I must be blind, because I re-checked both the manual and the product page and I could not find any reference to the 30 memory locations you're referring to.

steffenl82, 2019-04-08:
Sorry for the confusion! I checked the company's German homepage at

https://www.medisana.de/Gesundheitskontrolle/Personenwaagen/BS-444-connect-Koerperanalysewaage.html

half way down you will find "GEBRAUCHSANWEISUNG (PDF)" which links to

https://www.medisana.de/out/pictures/media/manual/40444_bs444connectwestru_09-2018_ver.1.4_web.pdf

see page 6

lbschenkel, 2019-08-05:
@oliexdev: FYI: I ended up buying a Medisana 444. Initially the multiple user functionality didn't work with openScale, it was syncing only one user, only the last measurement and I was also getting frequent errors. Because I was too busy at the time, I didn't play much with the scale or the app and just hoped that I could do more troubleshooting when I had the time.

Then I noticed the changelog for 2.1.0 ("replaced Bluetooth backend RxAndroidBle with Blessed Android library"), so I gave it another go. I'm happy to say that all the troubles I had went away, and not only it syncs perfectly but the history and multi-user functionality is working as well.

sebmate, 2020-03-28:
Hi all,

the problem still persists, at least for me (Android 7, Medisana BS444). As stated by steffenl82, the old version (<=1.9.3) correctly fetched all measurements, whereas the new version fetches only the last measurement. I kept switching between both versions to see if this got fixed many times, but as this is a pain I have decided to comment on this again.

Sometimes, the recent version fetches all or at least multiple measurements. I was not able to figure out when this happens. It might be after I have restarted the phone. 

What also bugs me is that it always tells me to "step barefoot on the scale" after it has fetched the measurement(s). Sometimes fetching does not work at all and I have to go to the Bluetooth settings menu in OpenScale and select the scale again.

I think I'll switch back to the old version again. Would be great if someone could have a look at this again. Anyway, MANY thanks to all developers for this useful app!

Sebastian