oliexdev/openScale #430: Exception on adding entry

wolkenschieber, 2019-03-06:
Following exception was thrown when I added an entry. App crashed afterwards 

'''
Build version: 2.0.1 
Build date: 1980-01-01 00:00:00 
Current date: 2019-03-06 07:10:39 
Device: Motorola Moto G Play 
 
Stack trace:  
android.view.WindowManager$BadTokenException: Unable to add window -- token android.os.BinderProxy@9687693 is not valid; is your activity running?
	at android.view.ViewRootImpl.setView(ViewRootImpl.java:697)
	at android.view.WindowManagerGlobal.addView(WindowManagerGlobal.java:342)
	at android.view.WindowManagerImpl.addView(WindowManagerImpl.java:93)
	at android.widget.Toast$TN.handleShow(Toast.java:459)
	at android.widget.Toast$TN$2.handleMessage(Toast.java:342)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:154)
	at android.app.ActivityThread.main(ActivityThread.java:6165)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:888)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:778)

'''

cmuerau, 2019-03-08:
Same here, ahppes in 95% of attempts when adding new entry. Device is in fact a Moto G4 Play under Android 7.1.1

Build version: 2.0.1
Build date: 1980-01-01 00:00:00
Current date: 2019-03-08 16:39:26
Device: Motorola Moto G Play

Stack trace:
android.view.WindowManager$BadTokenException: Unable to add window -- token android.os.BinderProxy@c6fce07 is not valid; is your activity running?
at android.view.ViewRootImpl.setView(ViewRootImpl.java:697)
at android.view.WindowManagerGlobal.addView(WindowManagerGlobal.java:342)
at android.view.WindowManagerImpl.addView(WindowManagerImpl.java:93)
at android.widget.Toast$TN.handleShow(Toast.java:459)
at android.widget.Toast$TN$2.handleMessage(Toast.java:342)
at android.os.Handler.dispatchMessage(Handler.java:102)
at android.os.Looper.loop(Looper.java:154)
at android.app.ActivityThread.main(ActivityThread.java:6165)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:888)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:778)

cmuerau, 2019-03-08:
Same here, happens in 95% of attempts when adding new entry. Device is in fact a Moto G4 Play under Android 7.1.1

Build version: 2.0.1
Build date: 1980-01-01 00:00:00
Current date: 2019-03-08 16:39:26
Device: Motorola Moto G Play

Stack trace:
android.view.WindowManager$BadTokenException: Unable to add window -- token android.os.BinderProxy@c6fce07 is not valid; is your activity running?
at android.view.ViewRootImpl.setView(ViewRootImpl.java:697)
at android.view.WindowManagerGlobal.addView(WindowManagerGlobal.java:342)
at android.view.WindowManagerImpl.addView(WindowManagerImpl.java:93)
at android.widget.Toast$TN.handleShow(Toast.java:459)
at android.widget.Toast$TN$2.handleMessage(Toast.java:342)
at android.os.Handler.dispatchMessage(Handler.java:102)
at android.os.Looper.loop(Looper.java:154)
at android.app.ActivityThread.main(ActivityThread.java:6165)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:888)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:778)

oliexdev, 2019-03-10:
please try out the openScale-dev version, it seems your Android version has a problem with the toast message see https://github.com/drakeet/ToastCompat Maybe it resolve we run it on the UI thread

cmuerau, 2019-03-10:
Downloaded Dev APK 2.0.1-dev_5a1a5f33, works for me without exception for at least 4 test entries in a row.

wolkenschieber, 2019-03-12:
Having seen this exception since I reported it. Maybe some hickup   after the update?
I don't think it's a device issue. It's a Moto g 4 play running untempered stock android. Should be close to a Google image avd.

cmuerau, 2019-03-12:
The current 2.0.1 (42) is still crashing on my device when adding an entry.
Strange detail: after writing the above, when I set the flag "Save error log" in the settings then the crashing stopped. I just wanted to autosave error report sand saw empty reports, but not longer constant crashes. 

Any ideas?

cmuerau, 2019-03-12:
The current 2.0.1 (42) is still crashing on my device when adding an entry.
Strange detail: after writing the above, when I set the flag "Save error log" in the settings then the crashing stopped. I just wanted to autosave error reports and find only empty reports, but not longer constant crashes.

Any ideas?

cmuerau, 2019-03-16:
I have to revert the entry above a little, as I made a figured out a new detail which I wasn't aware of during writing the abote posting.

When entering the weight in the start screen (in German: "Übersicht") or the diagram screen, crashs can e reproduced. If values are entered in the table view (in German: "Tabelle") no crash happens. 

Are this view so different behind the scenes?

closed 2019-03-17 by oliexdev