oliexdev/openScale #431: Sluggish app behaviour

wolkenschieber, 2019-03-07:
Since version 2.0 (versionCode ~42) the app is behaving sluggish. This means it takes considerably longer to start up and to refresh after adding an entry. Currently my app's database holds about 500 entries.

The effect might be worse for me, since I only have a low budget phone.

oliexdev, 2019-03-10:
turn off the rolling chart then it should be faster.

I have a problem with the refreshing of the rolling chart. Currently, I resolved it with recreating the whole activity, see
https://github.com/oliexdev/openScale/blob/5a1a5f339c6dacf5b43151c020f4e2ab27e89633/android_app/app/src/main/java/com/health/openscale/gui/fragments/OverviewFragment.java#L140

If you have a better solution, feel free to send a PR.

wolkenschieber, 2019-03-13:
Disabling rolling doesn't affect loading time. The only difference is that the overview shows all data instead of the recent one.

I'd expect the overview in rolling mode to only load visible data.

oliexdev, 2019-05-26:
please try out the latest dev version. I dynamically load the measurement data if the rolling chart is on, that should increase the performance.

closed 2019-06-02 by oliexdev

wolkenschieber, 2019-06-15:
Works great. Thank you.