oliexdev/openScale #432: Google Fit Synchronization

RichyHBM, 2019-03-07:
I'm swear there was a request to add this years ago, but I wanted to know if thoughts on this had changed. 

Google fit would provide an easy way to synchronise data to the cloud as well as allowing weight data in other apps.

If not something you would directly want to add, is it something you'd be open to adding if there were a PR?

oliexdev, 2019-03-10:
please read the FAQ, openScale will not support it directly for privacy and licence reasons.

closed 2019-03-10 by oliexdev

jottr, 2019-05-12:
@oliexdev so why did you publish an app on the play store for 2.99 EUR if it is a licensing/privacy issue? 
The privacy issue could easily be ameliorated by allowing the user to toggle Google Fit synchronisation...

To add to the confusion, there is a free version on F-Droid. It is unclear which one this is (the Lite version or the Pro version). 
On the play store there are three apps available: 

- openScale light
- openScale pro
- openScale sync

So, which of these apps offer which features? 
When should I purchase openScale sync? Only if I installed the light version? Or is this also required when using the pro or F-droid version?

oliexdev, 2019-05-12:
@jottr I don't know what to say everything you asked is written in the FAQ or is described in each version!? Do I really have to point every single question!?

jottr, 2019-05-13:
In the play store you you list the following under features for both the light and the pro app: 

> - Synchronize your weight to GoogleFit with openScale sync

I would assume that since you list it under "features" it would be a feature of the app. So is it? 
Or will I need to purchase the sync app as well? 

From the FAQ: 

> The openScale pro supports Bluetooth scales and the light version doesn't. But if you don't want to pay for the pro version, you could test or use the f-droid version, because the pro and f-droid version are identical.

Again, I am none the wiser. What do I need to do to get sync? 
Is this part of the pro app or not?

jottr, 2019-05-13:
Just read the ratings of the sync app on the app store. 
Proof is in the pudding:

[image]