oliexdev/openScale #433: Unable to connect to Sanitas SBF70 (GATT_INSUF_AUTHORIZATION)

engeeaitch, 2019-03-07:
**Describe the bug**
I am unable to create a Bluetooth connection to the Sanitas SBF70 scale since the latest update.

**To Reproduce**
Steps to reproduce the behavior:
1. Force Close App
2. Start App
3. Turn on Bluetooth
4. Toast message says connected, but then says disconnected.  Eventually an error is shown (see Debug Log).

Running 2.0.1 of the App on Android 8 (OnePlus 5)

**Expected behavior**
Expected to be able to connect and download readings.

**Debug log**
Attached.
[openScale_2019-03-07_22-20.01.txt](https://github.com/oliexdev/openScale/files/2943664/openScale_2019-03-07_22-20.01.txt)

oliexdev, 2019-03-10:
could you post the openScale log file if you have disable your GPS.

engeeaitch, 2019-03-11:
Hi - I have turned off Location (which I think turns off GPS) and tried to connect.  I've attached the file.  One more clue:  When I start Open Scale, I see the Spotify notification briefly in the Status Bar, and then it tells me that Spotify is using battery? Seems very strange?
[openScale_2019-03-11_20-52.txt](https://github.com/oliexdev/openScale/files/2954189/openScale_2019-03-11_20-52.txt)

jroth-haj, 2019-03-15:
Also Bluetooth problem sind Update to 2.0.1
For details please see attached log.
 
[openScale_2019-03-15_23-52.txt](https://github.com/oliexdev/openScale/files/2973179/openScale_2019-03-15_23-52.txt)

jroth-haj, 2019-03-16:
Got an older version 1.8.2 from my old phone.
In this version all is ok.

oliexdev, 2019-03-17:
I bought the SBF70 scale to try to reproduce this error but I can't reproduce it at all. On my SBF70 scale it works with no problems. I guess it some weird Bluetooth timing problems again!?

I did some timing changes, so please try out the [openScale-experimental version](https://github.com/oliedev/openScale/releases/tag/travis-dev_openScale-experimental-build) and attach the log file here.

engeeaitch, 2019-03-17:
Hi - I tried the experimental version - log attached. 
[openScale_2019-03-17_07-57.txt](https://github.com/oliexdev/openScale/files/2974917/openScale_2019-03-17_07-57.txt)

This version told me a couple of times, that the maximum number of users had been reached.  Looking at the logs, I can see that there are entries on the scale that I would want to delete.  Is there any way to do this? On re-trying the app reported a UUID error.

For info: I have tried the 2.0.1 app many times, and on just one occasion, I successfully connected to the scale, without changing anything as far as I was aware.

oliexdev, 2019-03-17:
@engeeaitch looking at your log files it looks not so bad. Did you try more then once to connect to the device? The first two attempts were successful regarding to your logs. And yes you have to many users on the scale. You can only delete the users if you reset the scale (push the button on the back of your scale when it is on. On the scale should then show up the message `del` for a short time)

engeeaitch, 2019-03-17:
@oliexdev Yes, I tried three times, but I was not successful in downloading a measurement.  The 2.0.1 version does not complain about too many users? (Thanks for the info about deleting users).

oliexdev, 2019-03-17:
@engeeaitch you only receive measurement if you measure yourself and then connecting to the scale afterwards. Please try this out and attach the log file here.

engeeaitch, 2019-03-17:
I have now downloaded my measurements using the experimental version - it was because I hadn't set up my date of birth correctly - I thought it only keyed on the Name.   So the experimental version seems to work better, but I am still getting the UUID error from time to time.

oliexdev, 2019-03-17:
@engeeaitch please try out the latest [openScale-experimental version](https://github.com/oliedev/openScale/releases/tag/travis-dev_openScale-experimental-build) and attach the log file if it works better.

jroth-haj, 2019-03-17:
Can't get the experimental Version to test it at my Pixel 3.

engeeaitch, 2019-03-17:
Works OK sometimes, but still getting the UUID error on other occasions.
[openScale_2019-03-17_12-32.txt](https://github.com/oliexdev/openScale/files/2975135/openScale_2019-03-17_12-32.txt)
When it connects successfully, the display of the scale turns on.  I don't remember this happening on other versions?

SriramKeerthi, 2019-03-20:
Works when I turn off location. Fails when I enable location:

```
2019-03-20 09:42:06.916 Debug [11337] RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(169378142) in 68 ms
2019-03-20 09:42:06.917 Debug [11324] RxBle#BluetoothGatt$1: onCharacteristicChanged characteristic=0000ffe1-0000-1000-8000-00805f9b34fb
2019-03-20 09:42:06.928 Debug [11329] BluetoothCommunication: Write characteristic 0xffe1: E7 F1 59 03 02
2019-03-20 09:42:06.930 Debug [11335] BluetoothCommunication: onCharacteristicChanged 0xffe1: E7 59 03 03 9E 01 58 01 1E 08 85 09 CC 01 34
2019-03-20 09:42:06.932 Error [11335] BluetoothBeurerSanitas: java.lang.NullPointerException: Attempt to read from field 'int com.health.openscale.core.bluetooth.BluetoothBeurerSanitas$RemoteUser.localUserId' on a null object reference
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.processMeasurementData(BluetoothBeurerSanitas.java:350)
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.processMeasurement(BluetoothBeurerSanitas.java:401)
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.onBluetoothNotify(BluetoothBeurerSanitas.java:282)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.lambda$setNotificationOn$15(BluetoothCommunication.java:349)
	at com.health.openscale.core.bluetooth.-$$Lambda$BluetoothCommunication$69nMflHUCojTBR8vbvCSdwRStSc.accept(Unknown Source:6)
	at io.reactivex.internal.observers.LambdaObserver.onNext(LambdaObserver.java:63)
	at io.reactivex.internal.operators.observable.ObservableRetryPredicate$RepeatObserver.onNext(ObservableRetryPredicate.java:69)
	at io.reactivex.observers.SerializedObserver.onNext(SerializedObserver.java:111)
	at io.reactivex.internal.operators.observable.ObservableDelay$DelayObserver$OnNext.run(ObservableDelay.java:114)
	at io.reactivex.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:66)
	at io.reactivex.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:57)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.lang.Thread.run(Thread.java:764)

2019-03-20 09:42:31.849 Debug [2] AboutPreferences: Debug log disabled
```

oliexdev, 2019-03-20:
@SriramKeerthi please attach the whole log file and which version did you try?

giuaig, 2019-03-20:
I have the same status 8 error message with v2.0.1 from FDroid, both with gps on or off. Scale sbf70. It was ok with previous openScale version

[Openscale-GATT_INSUF_AUTHORIZATION.txt](https://github.com/oliexdev/openScale/files/2990220/Openscale-GATT_INSUF_AUTHORIZATION.txt)

reklawykSekuL, 2019-03-24:
**@oliexdev**  testeted the dev version  with my BF710: seems to work more stable now with the new bt-library wersion. tested with  GPS off an on
With GPS off:
[openScale_2019-03-24_10-45.txt](https://github.com/oliexdev/openScale/files/3000251/openScale_2019-03-24_10-45.txt)

With GPS on:
[openScale_2019-03-24_10-53.txt](https://github.com/oliexdev/openScale/files/3000254/openScale_2019-03-24_10-53.txt)

oliexdev, 2019-03-24:
should be more stable with openScale v2.0.2 (but be aware it is a timing problem of the scale because the scale manufacture did some faulty Bluetooth implementation on the scale)

closed 2019-03-24 by oliexdev