oliexdev/openScale #435: Unable to sync with SilverCrest SBF75 after not syncing for some time

danielfaust, 2019-03-11:
I had noticed that it was possible to take weight measurements without having the phone connected to the scale, and that it would then download previously stored measurements once the app connected with the scale.

This can be reproduced as follows. Take measurements with the scale every day for about 1-2 weeks without using the app to sync. Then enable the BT-Button in the app to start a sync process.

The app confirms the connection to the scale, but then outputs two toasts, one which can't be read because it is immediately replaced by another one stating:

> Unexpected Bluetooth-Error: Cannot set local notification (code 1) with characterisitc UUID `0000ffe1-0000-1000-8000-00805f9b34fb`"

Then it won't sync.

Here's the debg log.

[openScale_2019-03-11_11-03.txt](https://github.com/oliexdev/openScale/files/2951681/openScale_2019-03-11_11-03.txt)

oliexdev, 2019-03-11:
could you post the openScale log file if you have disable your GPS.

danielfaust, 2019-03-11:
Disabling GPS made it work, it synced the last couple of measurements (I think the device has a buffer of 6 measurements)

How strange, right?

Here's the full debug log.

[openScale_2019-03-11_20-23.txt](https://github.com/oliexdev/openScale/files/2953922/openScale_2019-03-11_20-23.txt)

----

But I just checked if it now works again with GPS enabled, while taking direct measurements, and for some reason now it isn't working at all, regardless of the GPS setting.

This time I also got a toast telling me that I cannot create more users, which is somewhat odd. Maybe the reason was that now I tried it with full clothing on (Jeans and Jacket), maybe it was too much of a weight change (also with socks, so that no fat could be measured)? Search for "BluetoothBeurerSanitas: Create user: test" in the following log

[openScale_2019-03-11_20-33.txt](https://github.com/oliexdev/openScale/files/2953936/openScale_2019-03-11_20-33.txt)

oliexdev, 2019-03-11:
could you try out the [openScale-experimental version](https://github.com/oliexdev/openScale/releases/tag/travis-dev_openScale-experimental-build) and could you also attach the log file of this version?

danielfaust, 2019-03-11:
It doesn't fix the issue :(

It adds more problems by not keeping the BT-enabled button in sync (greyed out while establishing connection and when connection is made)

[openScale_2019-03-11_22-58.txt](https://github.com/oliexdev/openScale/files/2954392/openScale_2019-03-11_22-58.txt)

oliexdev, 2019-03-12:
please try again the latest openScale-experimental version. I added some delays maybe it's works better!?

weliem, 2019-03-13:
It looks like the writing of the init command times out after exactly 5 seconds. Is RxAndroidBLe doing that or is it the phone itself?

danielfaust, 2019-03-13:
I think it will make more sense if I try to debug it by myself. I have experience with BLE on Android (but not with normal Bluetooth where you need to establish a real connection) and think I could figure it out.

My only problem is the following: I don't want to replace the version I downloaded from Google Play, but install this experimental version in parallel, just like I was able to do with the APK which you provided to me.

I downloaded the experimental version and compiled it, but I didn't install it on the device because I'm afraid that I could mess up my profiles and stored data from the Google Play version.

How do I proceed? How did you manage to make it getting installed in parallel without changing the package name `com.health.openscale`?

danielfaust, 2019-03-13:
Ok, so I changed the `applicationId` in the build.gradle as well as the `versionName` and installed it on another phone which already had OpenScale installed (my previous device), and it installed it side-by-side as desired.

But something strange happened, when I connected it synced, but it downloaded data from January / February of 2018.

The last measurements from this "old" phone in the official OpenScale app are from November 2018, which is OK, that's when I switched phones.

I assume that there were some old values stored in the scale, maybe as another user profile, even tough I configured the user in the app with the same name "test" in both devices and in both versions.

The official app on the old phone has one measurement from January 2018, then none until July 2018, I think because at that time I used the Libra app and was skipping the BT functionality. I bought the scale in January 2018 with the intention to have a Python script (RasPi Zero W) read out the values and store them in a Database.

Those measurements from Jan/Feb 2018 in this old version must be from another profile, when I was doing the Python testing. 18 measurements were downloaded from the scale.

In any case, I think I'm now set up to do some debugging.

danielfaust, 2019-03-13:
So i just installed it in the new device, and when I tried to sync the BT problem was present again. Then back with the old one I got the message with "too many users" or something like that.

I installed the official app (HealthForYou) to see if this can help, and it turned out that there are only two profiles on the scale. I deleted one which had a wrong birthdate on it, and the other one had still 20 measurements on it which I deleted with the app.

I think this app (OpenScale) could benefit of some kind of checkbox to disable the autocreation of users, that would move one possible problem out of the way.

danielfaust, 2019-03-13:
I think this is related to the cleanup of the BT-logic performed when a disconnection takes place. When I get out of range or press the BT-button in the app (the one next to the +) in order to "force" a disconnection, then the cleanup is started (possibly all that follows `Connection operations queue to be terminated` until `RxBle#ClientOperationQueue: FINISHED DisconnectOperation`).

Starting from that moment, the app won't be able to sync again, then it will throw the `GATT_INSUF_AUTHORIZATION` error after a new connection has been started (don't know yet if the connection is successful, but the Toast indicates that yes).

When I now force close the app (go into the App-Details Activity which has the Uninstall, Force Close, Permissions...) and then connect it again, then the connection and all the communication is successful until a disconnect occurs (out of range, disconnect button, haven't tested a timeout which I assume exists). From that point on it doesn't work until I force close the app.

Toggling GPS off or on instead of force closing the app does not fix the issue. I don't know why turning it off worked a couple of days ago, maybe the app got killed because of the GPS UI.

GPS really appears to be unrelated to this. I've tried this several times with different GPS settings.

Sometimes it also fails after a force close, but it almost always fixes the issue.

danielfaust, 2019-03-13:
Nope. I took one weight measurement and now can't sync again. From Google Play won't work at all, the experimental also doesn't work. Also it turns out that I deleted the wrong user, since on the new phone I had the wrong birthday and on the old one the correct one.

I think I' moving over to Libra and try to implement this in Python, which was my original plan. Maybe if I find some time for this I will try to find out what the problem is in this app, but it's hard to tell what's wrong with it.

Can I reassign the user by using this Intelligent User Asignment? (I have the German UI, sorry)

oliexdev, 2019-03-17:
I will close this issue because it is highly related to issue #433 
Please comment on this issue for further related problems.

closed 2019-03-17 by oliexdev