oliexdev/openScale #436: How about trendlines

Gnopps, 2019-03-13:
Would it be possible to add trendlines like the app Libra has? Trendlines make for an easier and (in my opinion) more interesting view of the weight chart.

[image]

oliexdev, 2019-03-17:
which line do you mean in the above graph?
How do you calculate your trend line?

naoliv, 2019-03-18:
The small circles are weight measurements, blue line is the goal line and the strong red line is a smoothed curve calculated from the the previous measurements.
The trend line is the light red line (in this case, based on previous measures, the tendency is for the person to continue to lose weight)

oliexdev, 2019-03-18:
@naoliv thanks but how the trendline calculated? Do you have a formula?

Gnopps, 2019-03-18:
It is an exponential moving average of 7 days based on the hacker's diet. I'm no mathematician so can't explain better than that. :-)

Some links: 
http://www.fourmilab.ch/hackdiet/e4/signalnoise.html
https://en.wikipedia.org/wiki/Moving_average#Exponential_moving_average

naoliv, 2019-03-18:
From what I saw they use this https://www.fourmilab.ch/hackdiet/comptools.html which lead us to https://www.fourmilab.ch/hackdiet/online/hdo.html#Trend and then to this part https://www.fourmilab.ch/hackdiet/e4/pencilpaper.html (there is a `Calculating best fit trends` section at the end; maybe some other parts could be interesting for openScale too)

goldfndr, 2019-04-05:
For whatever it's worth, Libra is also configurable for values other than 7:

* Count of days for smoothing the past/present (which it calls the trend, I think Hacker's Diet calls it that too).
* Count of days for forecasting the future (which it calls the forecast).

The original EatWatch app for Palm (based on the Hacker's Diet concepts, written by the same Autodesk guy) didn't provide a forecast.

oliexdev, 2019-05-26:
As I understand they use a simple moving regression line for the trend analysis. I added linear regression lines for the graph views not for the overview graph because with rolling chart on the linear regression line destroys the viewport if you scroll back in time.
You get the average line if you select another view mode (for example week, month or year) 
Please test it and give feedback. What do you think?

herm, 2019-07-21:
The trend line in Libra is **not** a polynomial regression. It might not be clear from the screenshot @Gnopps posted, but the trend line is the red line. The blue line is the target line. The light red line is a prediction of your weight in the future.

The red line is an exponential moving average. The advantage is that you do not need a mathematical model for your data. It is just an average of all measurements with more weight given to more recent measurements.

On the other hand for a fit to make sense you need a model of your data. While a linear model is somewhat reasonable (constantly gaining or losing weight) I can't imagine how a higher order polynomial (e.g. weight = A * time² + B * time) might make any sense. 

I would propose to behave exactly like "Libra". It is one of the best representations I have ever seen for a weight tracking app. That means: Use exponential moving average for the trend line. Use a linear fit for estimating future values (but start the estimate at the value of the trend line, not at the value of the last measurement). Instead of a fixed target allow defining a start and end date and weight for your target. In combination with the trend line you can see at a glance if you are likely to reach your target in time.

herm, 2019-09-11:
@oliexdev Are there any plans to implement trend lines like I described? If so could you please reopen this issue to track the progress?

oliexdev, 2019-11-05:
@Marwel What do you mean with artifacts?

ati3, 2019-12-10:
I would love this feature and I know how to implement it. Unfortunately, I don't have time to study your code and implement the feature but I think I can explain how to do it.   So, if someone needs help with something, just ask.

oliexdev, 2019-12-12:
@ati3 maybe you could help @Marwel how is the progress in the implementation?

Marwel, 2019-12-13:
None so far. Didn't have the time to fiddle with it. Maybe over the holidays, when family is not bugging me for tech support.

ati3, 2019-12-13:
I can write some methods to take care of the math if it helps. I just need input objects and return values so you can use the methods later.

Marwel, 2020-01-16:
So i finally had some time to play with it, but i'm stuck on several topics.
I took the code of the regression line and tried to modify it, so that it draws a line with some changed values, which works fine. But i struggle with linedataSets, which contains more than one line of measurements and i fail to get "historical" data.
The first problem gives me more lines than i want, pe i get a line for the measurements, one for goal and then for both of them the "hackdiet" one.
The second problem requires some more time on my side. I'll have to understand more of the code to know where to get the measurements of the last week to correctly calculate the "hackdiet" entry.

ati3, 2020-01-17:
This is a big change, not just replacing one line with other. There are 4 things to calculate before drawing those 3 curves, and they all require values from the past (ideally all, but at least 1 or 2 months). I would suggest to divide it to small changes.
The current app works in a different way: it shows 1 month and calculate everything based on that. This is a good start, but the new trendlines require all the values. The code may not be prepared for this. Btw this feature would be more natural with graphs that can be continuously slided to the past. Maybe it would be better to start with that, since it is easier and prepares the code for next steps.
One more thing: the new trendlines must go to a separate graph, the existing one is already too crowded with %s, kgs etc.

oliexdev, 2020-01-19:
@Marwel Did you have a public repository with your current status? The origin data is `scaleMeasurementList` which are transferred into a `lineDataSet` depending on your selection (weeks, months, day, years).

@ati3 

> There are 4 things to calculate before drawing those 3 curves, and they all require values from the past (ideally all, but at least 1 or 2 months). 

I intentionally don't load all history data because of issue #431. If you have large data sets the app will be sluggish. 

> The current app works in a different way: it shows 1 month and calculate everything based on that. This is a good start, but the new trendlines require all the values

that's not true. You can select exactly if you want to show the month, years, day, weeks. Only the needed values are fetched and shown.

> Btw this feature would be more natural with graphs that can be continuously slided to the past. 

You can already do that in the overview page (option rolling chart needs to enabled). Again you can select the spacing between the data points. The data entry are then the average value for e.g. months.

> One more thing: the new trendlines must go to a separate graph, the existing one is already too crowded with %s, kgs etc.

That's why you can hide/show every line using the measurement bar or you can configure on which axis bar should the line belong to in the settings.

Marwel, 2020-01-20:
@ati3 I don't believe, that i'm the right one for big changes. My Java knowledge is far too thin for that.
@oliexdev No, this isn't something, that i would be proud of sharing so far, because it's only copy and paste up to now.
I agree, that the overview page would be a good place for this and it should only be an option to enable or disable. The start would be to just show the hackdiet calculated line and then the trendlines. After that, we could argue about the look.

@oliexdev Thanks for the info.
I'll try to get this working in the overview.

oliexdev, 2020-07-05:
I added a trend line like in the Hacker's Diet, please try out the latest dev build and give feedback.

herm, 2020-07-18:
Is there an APK available somewhere?

oliexdev, 2020-07-26:
no just a branch repo

oliexdev, 2020-08-16:
@all I have a first implementation, now I need beta tester for this feature, see #612

closed 2020-09-13 by oliexdev