oliexdev/openScale #440: Can't find body metric estimations

naoliv, 2019-03-18:
**Describe the bug**
This wiki page https://github.com/oliexdev/openScale/wiki/Body-metric-estimations says openScale has some metric estimations implemented (body fat, water and lean mass) but I can't find this info anywhere.

In Statistics I can't see any this info, for example:

[image]

From the wiki page I would expect to have the masses calculated by only having my age, weight, height and sex.

**To Reproduce**
I don't know exactly where I should be seeing this kind of information that I am looking for, but:
1. Go to Statistics
2. See that there is nothing about body water, fat, etc (as can be seen at the screenshot above)

I am using Google Play's `v2.0.1 (42)`

**Expected behavior**
I should be seeing the calculated masses somewhere in the app.

**Debug log**
What I have in the debug log:

```
2019-03-18 15:25:45.169 Debug [2] AboutPreferences: Debug log enabled, openScale v2.0.1 (42), SDK 28, Xiaomi Mi A2 Lite
2019-03-18 15:25:45.172 Debug [2] AboutPreferences: Selected user id(1) name(N) birthday(Thu Apr 28 00:00:00 GMT-03:00 1983) age(35) body height(163.00) scale unit(kg) gender(male) initial weight(81.40) goal weight(74.00) goal date(Sat Aug 24 23:57:41 GMT-03:00 2019) measure unt(cm) activity level(0)
```

oliexdev, 2019-03-18:
Go to settings->measurements then click on body fat for example and turn on the estimation.

closed 2019-03-18 by oliexdev

naoliv, 2019-03-18:
Oh, right. Thanks a lot!