oliexdev/openScale #442: Xiaomi Mi scale v2 history support

Lurker00, 2019-03-20:
Wiki says, history data in Xiaomi Mi scale v2 is not available. But I have this device, and I'm able to download historical measurements from it, e.g. using Notify & Fitness Amazfit application. I've tried it on two different Android devices, so I'm confident in this statement.

How can I help you to add history support for Scale 2? N&F does not have multi-user support, Mi Fit is not reliable with history, so, currently, openScale is my only hope!

I'm a programmer, also with some experience with Android.

As I understand, openScale supports history for v1. From your source code, I see v2 misses onBluetoothRead method, and looks identical in every other aspect (but data format, ofcourse!).

ivoid, 2019-03-21:
I have the Mi Scale v2 and openScale does support history. In fact, I wish there's a way to disable history since every time I need to reinstall openScale for some reason it reads all of my history again.

oliexdev, 2019-03-24:
@Lurker00 it is fixed. The mi scale v1 and v2 are identical beside that the v2 supports body measurements. It is also possible to download the history data of the mi scale but you have to first time register on the scale afterwards all measurements are downloaded

@ivoid yes openScale download all measurements again because no data on the scale is ever deleted. You could export you database via backup and import it. Then previous marked deleted entries in openScale are not imported again from mi scale.

closed 2019-03-24 by oliexdev

deeeed, 2021-02-24:
> @Lurker00 it is fixed. The mi scale v1 and v2 are identical beside that the v2 supports body measurements. It is also possible to download the history data of the mi scale but you have to first time [...]

Could you elaborate what you mean by "first time register on the scale afterwards"?
Currently I can only get my latest measurements from Xiaomi v2. Is there a way to reset and re-download all values from the scale?