oliexdev/openScale #443: Backup incomplete: does not save Users or Bluetooth scales

lolilolicon, 2019-03-25:
What is the backup backing up anyway? Looks like it's the same info as exported .csv? Namely, only measurements?

I tried restoring from a backup, expecting to get my users and settings back. But no, openScale does not restore these at all. So am I supposed to set up all my users and settings all over again?

Please make the backup back up the complete data, so that a restore will actually restore everything.
Thanks!

lolilolicon, 2019-03-25:
Hmm, I sense something else's up. Namely Android 9 journaled database. See [this issue](https://github.com/federicoiosue/Omni-Notes/issues/653#issuecomment-458290332) from another project.

oliexdev, 2019-03-25:
I don't like your tone @lolilolicon it is an open source app, feel free to contribute instead of complaining. 
Also please write a proper bug report using the template and attach the openScale log file.

lolilolicon, 2019-03-25:
Sorry, the opening post wasn't very polite. I hope you understand, I was quite frustrated.

As I explained in my second post, I believe the issue only occurs on Android 9, which I recently updated to. The same issue caused backup restore issue for Omni Notes, so I'm twice bitten here.

So just to confirm: backup is complete on older versions of Android, correct?
If so, I'll update the issue title properly.

As you requested the log, I wiped all openScale data, started over, added a user, enabled logging and successfully triggered the error as reported in post 2. Unfortunately, the log has only a few lines about my newly added user, and nothing about error. I'm using openScale v2.0.1 (42) btw.

Again, I'm rather confident this is caused by the "journaled database" on Android 9.
Please look into it. If you need my assistance testing, I'm at your service.

oliexdev, 2019-03-26:
No, it is only currently backup your measurement database. The user database is not backup. How many users do you have in openScale? How much effort is it to generate a user?

The journaled database is disabled during the exporting / importing the database which is fixed in issue #344 

Which openScale version did you create the backup and in which version do you try to import it?

lolilolicon, 2019-03-26:
> No, it is only currently backup your measurement database. The user database is not backup.

That's unfortunate, IMHO. For measurements we already have the **.csv** _export_ function (which is good for what it does).
In my understanding, a _backup_ should be more complete. Ideally, the user should be be able to export a backup, wipe the app data, import the backup, and restore everything -- as if nothing happened at all.

> How many users do you have in openScale? How much effort is it to generate a user?

In my moderate usage, three users. Not a LOT, but there's some effort.
What's more, I have to click around and restore all my settings (which measurements to show or hide, etc.), re-search the Bluetooth scale & remember to recheck some more options.
I think this is really unnecessary friction.

> The journaled database is disabled during the exporting / importing the database which is fixed in issue #344

Very cool!

> Which openScale version did you create the backup and in which version do you try to import it?

Both openScale v2.0.1. In fact, I tested the backup & restore in the same app on the same Android 9 device.
I tried it again:
1. Settings - Backup - Export backup
2. Settings - Measurements - Delete all
3. Settings - Backup - Import backup

I did this twice. The first time it's an error. The second it succeeds (So it kinda works). Nothing reported in log.
I tested this with only one user & one's user's measurements.

Now, since only measurements are backed up, how does restore work? Say the backup includes 3 users' data, and now I have only added one user in my newly installed app, what measurements are restored upon backup import? Regardless. I hope these questions will become a moot point -- as I hope you will consider making the backup complete.

oliexdev, 2019-03-27:
> That's unfortunate, IMHO. For measurements we already have the .csv export function (which is good for what it does).

with the .csv export you get only the current user measurement with the backup you get all measurements all the measurements that internal marked as deleted.

>  In my understanding, a backup should be more complete. Ideally, the user should be be able to export a backup, wipe the app data, import the backup, and restore everything -- as if nothing happened at all.

sure but the effort to implement this behaviour is huge compared to manual add some users (most users have not more users than 2-3) but you are welcome to implement this.

> Both openScale v2.0.1. In fact, I tested the backup & restore in the same app on the same Android 9 device.
> I tried it again:
> 
>     Settings - Backup - Export backup
>     Settings - Measurements - Delete all
>     Settings - Backup - Import backup
> 
> I did this twice. The first time it's an error. The second it succeeds (So it kinda works). Nothing reported in log.

I can't reproduce this error and I tried it multiply times.

> Now, since only measurements are backed up, how does restore work? Say the backup includes 3 users' data, and now I have only added one user in my newly installed app, what measurements are restored [...]

all measurements are restored! The backup is the whole copy of the sql measurement database. The link to a user is the userId.

lolilolicon, 2019-03-27:
> all measurements are restored! The backup is the whole copy of the sql measurement database. The link to a user is the userId.

OK, that's good to know.

So let's consider the standard procedure for backup & restore...
To import a backup, the user must first create as many users as there were when the backup was exported, and in the _exact same order_, too. Only then, import the backup.
Is that correct?

Having thought through this procedure, my conclusion is, well, might as well forget about backup.
It's less error prone to just use export .csv.
Compared to recreating 3 users, what's clicking a few times to export 3 users' data separately?

Simply put, the backup feature as it stands is useless, and misleading.

oliexdev, 2019-03-27:
> So let's consider the standard procedure for backup & restore...
> To import a backup, the user must first create as many users as there were when the backup was exported, and in the exact same order, too. Only then, import the backup.
> Is that correct?

yes

> 
> Having thought through this procedure, my conclusion is, well, might as well forget about backup.
> It's less error prone to just use export .csv.
> Compared to recreating 3 users, what's clicking a few times to export 3 users' data separately?

sure you can do it also that way.

> Simply put, the backup feature as it stands is useless, and misleading.

no it is not useless because as I wrote it saves the whole database but yes in openScale you have both ways to backup your data and a third if you are using openScale sync to sync it with GoogleFit.

closed 2019-03-27 by oliexdev

gappleto97, 2019-03-29:
At the very least, why is this not documented in the backup settings? I would not have known this from using the app, and I dont know how I was supposed to know it.

oliexdev, 2019-03-29:
@gappleto97 well write a doc and I can put it on the wiki. But as you can see most people are unable to read the FAQ not the GooglePlay description. Beside also you can improve the backup mechanism, send a PR with your better implementation.