oliexdev/openScale #445: Unable to connect to Sanitas SBF70 (See issue #433)

engeeaitch, 2019-03-26:
@oliexdev Hi - I have downloaded version 2.0.2 but I am now unable to get any connection to my scale.
 I have tried with Location on and off.  Please see the debug log attached.
[openScale_2019-03-26_22-26.txt](https://github.com/oliexdev/openScale/files/3010686/openScale_2019-03-26_22-26.txt)

oliexdev, 2019-03-27:
but you wrote in issue #433 that it is works sometimes https://github.com/oliexdev/openScale/issues/433#issuecomment-473661667
It is a timing problem more I cannot do, it is on the scale side.

closed 2019-03-27 by oliexdev

engeeaitch, 2019-03-27:
When I tried yesterday, it failed four out of four times.  I've tried tonight, and it worked twice, followed by six failures, followed by one success.  Is there nothing that can be done to improve the chances of success?

runyalfa, 2019-03-28:
I tried with the 2.0.3 version and in my case it's impossible to connect....

engeeaitch, 2019-04-03:
@oliexdev Would you take another look at this please - it's taking around 5 minutes to be successful in getting a connection - I would be better off writing the results down on a piece of paper and entering them manually.  Appreciate that timing issues can be difficult - but the earlier versions worked just fine.
P.S. Have paid for the Pro version in the Play Store.