oliexdev/openScale #446: App crashes when started

Irihapeti, 2019-03-27:
Latest version of openScale crashes when loaded. OS is LineageOS 14.1

Error message follows.
-----

Build version: 2.0.2 
Build date: 1980-01-01 00:00:00 
Current date: 2019-03-27 23:46:13 (UTC +13)
Device: Motorola Moto G Play 
 
Stack trace:  
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.health.openscale/com.health.openscale.gui.MainActivity}: java.lang.NullPointerException: Attempt to invoke interface method 'android.view.MenuItem android.view.MenuItem.setChecked(boolean)' on a null object reference
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2684)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2751)
	at android.app.ActivityThread.-wrap12(ActivityThread.java)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1496)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:154)
	at android.app.ActivityThread.main(ActivityThread.java:6186)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
Caused by: java.lang.NullPointerException: Attempt to invoke interface method 'android.view.MenuItem android.view.MenuItem.setChecked(boolean)' on a null object reference
	at com.health.openscale.gui.fragments.OverviewFragment.onCreateView(OverviewFragment.java:160)
	at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2440)
	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:885)
	at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1229)
	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:1295)
	at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:686)
	at androidx.fragment.app.FragmentManagerImpl.executeOps(FragmentManagerImpl.java:2057)
	at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManagerImpl.java:1847)
	at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManagerImpl.java:1802)
	at androidx.fragment.app.FragmentManagerImpl.execPendingActions(FragmentManagerImpl.java:1709)
	at androidx.fragment.app.FragmentManagerImpl.dispatchStateChange(FragmentManagerImpl.java:2609)
	at androidx.fragment.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManagerImpl.java:2565)
	at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:245)
	at androidx.fragment.app.FragmentActivity.onStart(FragmentActivity.java:525)
	at androidx.appcompat.app.AppCompatActivity.onStart(AppCompatActivity.java:179)
	at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1249)
	at android.app.Activity.performStart(Activity.java:6701)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2647)
	... 9 more

michaelprager, 2019-03-27:
Same issue here. Interesting build date btw.

Build version: 2.0.2 
Build date: 1979-11-30 01:00:00 
Current date: 2019-03-27 12:52:16 
Device: Google Pixel 2 XL 
 
Stack trace:  
java.lang.NullPointerException: Attempt to invoke interface method 'android.view.MenuItem android.view.MenuItem.setChecked(boolean)' on a null object reference
	at com.health.openscale.gui.fragments.OverviewFragment.onCreateView(OverviewFragment.java:160)
	at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2440)
	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:885)
	at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1229)
	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:1295)
	at androidx.fragment.app.FragmentManagerImpl.dispatchStateChange(FragmentManagerImpl.java:2605)
	at androidx.fragment.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManagerImpl.java:2565)
	at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:245)
	at androidx.fragment.app.FragmentActivity.onStart(FragmentActivity.java:525)
	at androidx.appcompat.app.AppCompatActivity.onStart(AppCompatActivity.java:179)
	at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1391)
	at android.app.Activity.performStart(Activity.java:7165)
	at android.app.ActivityThread.handleStartActivity(ActivityThread.java:2975)
	at android.app.servertransaction.TransactionExecutor.performLifecycleSequence(TransactionExecutor.java:180)
	at android.app.servertransaction.TransactionExecutor.cycleToPath(TransactionExecutor.java:165)
	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:142)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:70)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1816)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:193)
	at android.app.ActivityThread.main(ActivityThread.java:6718)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)

woj-tek, 2019-03-27:
Exactly the same issue.

Btw. very nice screen with info where to report the issue, however - would it be possible to make the URL into a link?

bushrang3r, 2019-03-27:
Same here.

Android 9.0 
Resurrection Remix Rom
Galaxy NOTE 4

Build version: 2.0.2 
Build date: 1979-11-30 09:00:00 
Current date: 2019-03-27 21:47:51 
Device: Samsung SM-N910F 
 
Stack trace:  
java.lang.NullPointerException: Attempt to invoke interface method 'android.view.MenuItem android.view.MenuItem.setChecked(boolean)' on a null object reference
	at com.health.openscale.gui.fragments.OverviewFragment.onCreateView(OverviewFragment.java:160)
	at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2440)
	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:885)
	at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1229)
	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:1295)
	at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:686)
	at androidx.fragment.app.FragmentManagerImpl.executeOps(FragmentManagerImpl.java:2057)
	at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManagerImpl.java:1847)
	at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManagerImpl.java:1802)
	at androidx.fragment.app.FragmentManagerImpl.execPendingActions(FragmentManagerImpl.java:1709)
	at androidx.fragment.app.FragmentManagerImpl.dispatchStateChange(FragmentManagerImpl.java:2609)
	at androidx.fragment.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManagerImpl.java:2565)
	at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:245)
	at androidx.fragment.app.FragmentActivity.onStart(FragmentActivity.java:525)
	at androidx.appcompat.app.AppCompatActivity.onStart(AppCompatActivity.java:179)
	at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1391)
	at android.app.Activity.performStart(Activity.java:7165)
	at android.app.ActivityThread.handleStartActivity(ActivityThread.java:2976)
	at android.app.servertransaction.TransactionExecutor.performLifecycleSequence(TransactionExecutor.java:180)
	at android.app.servertransaction.TransactionExecutor.cycleToPath(TransactionExecutor.java:165)
	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:142)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:70)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1817)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:193)
	at android.app.ActivityThread.main(ActivityThread.java:6723)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:495)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)

kelos, 2019-03-27:
More
Build version: 2.0.2 
Build date: 1980-01-01 00:00:00 
Current date: 2019-03-27 12:44:24 
Device: HUAWEI CAM-L03 
 
Stack trace:  
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.health.openscale/com.health.openscale.gui.MainActivity}: java.lang.NullPointerException: Attempt to invoke interface method 'android.view.MenuItem android.view.MenuItem.setChecked(boolean)' on a null object reference
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2479)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2539)
	at android.app.ActivityThread.access$900(ActivityThread.java:168)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1378)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:150)
	at android.app.ActivityThread.main(ActivityThread.java:5659)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:822)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:712)
Caused by: java.lang.NullPointerException: Attempt to invoke interface method 'android.view.MenuItem android.view.MenuItem.setChecked(boolean)' on a null object reference
	at com.health.openscale.gui.fragments.OverviewFragment.onCreateView(OverviewFragment.java:160)
	at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2440)
	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:885)
	at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1229)
	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:1295)
	at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:686)
	at androidx.fragment.app.FragmentManagerImpl.executeOps(FragmentManagerImpl.java:2057)
	at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManagerImpl.java:1847)
	at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManagerImpl.java:1802)
	at androidx.fragment.app.FragmentManagerImpl.execPendingActions(FragmentManagerImpl.java:1709)
	at androidx.fragment.app.FragmentManagerImpl.dispatchStateChange(FragmentManagerImpl.java:2609)
	at androidx.fragment.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManagerImpl.java:2565)
	at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:245)
	at androidx.fragment.app.FragmentActivity.onStart(FragmentActivity.java:525)
	at androidx.appcompat.app.AppCompatActivity.onStart(AppCompatActivity.java:179)
	at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1241)
	at android.app.Activity.performStart(Activity.java:6392)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2442)
	... 9 more

oliexdev, 2019-03-27:
thanks should be fixed with commit https://github.com/oliexdev/openScale/commit/3088633dc52f6d625317a2b2fe819038adb197f3

oliexdev, 2019-03-27:
> Btw. very nice screen with info where to report the issue, however - would it be possible to make the URL into a link?

we already asked the library to support this but currently this is not supported.

closed 2019-03-27 by oliexdev

gappleto97, 2019-03-28:
Any ETA on when this will hit FDroid? It wont let me downgrade the app.

oliexdev, 2019-03-29:
it usually takes 2-3 days up to a week for the f-droid bot

Irihapeti, 2019-03-29:
You may be able to downgrade the app by using this method: https://www.xda-developers.com/downgrade-an-app-android-no-root/

gappleto97, 2019-03-30:
That doesnt work on newer Android versions, sadly

gappleto97, 2019-03-30:
https://android.googlesource.com/platform/frameworks/base/+/921dd75