oliexdev/openScale #447: Rasp Pi receiver

pksorensen, 2019-03-27:
**Describe the solution you'd like**
I was looking to reverse engineer my Beurer BF700 scale to send weights to an endpoint of mine to plot my weight on the web.

Then I found this project and was thinking maybe it's worth going a bit more general.

Would it be possible to make a simple application that we could deploy/install on a rasp pi that sends information to an endpoint that was configurable?

If this is interesting, I could provide a simple web service that could plot the graphs. I also dont mind coding the application for the rasp pi if someone from this project could help me get started.

Basically, i want to plug in a rasp pi to power near the scale and every time someone steps on the scale, it sends the information to a configurable endpoint (HTTP POST)

lenucksi, 2019-03-27:
There is this [code](https://github.com/keptenkurk/BS440), maybe it would be worth to contribute there using the reverse engineered insights about the scales researched in this project.

oliexdev, 2019-03-27:
@pksorensen sure you will find the openScale implementation for the Beurer BF700 scale at https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothBeurerSanitas.java

But please aware that you have to comply with the GPLv3! (your source code must be also under the GPLv3!) Note the code which @lenucksi link is under the MIT, you can mixed them but the result must be again the GPLv3.

closed 2019-03-27 by oliexdev